ROBOTIS-GIT/dynamixel-workbench #9: VelocityControl Tuturial rosservice crash with DXL 1.0

airduke, 2016-10-16:
> "rosservice call /dynamixel_workbench_tutorials?wheel 0 0"

 (or any two values) crashed the service call and the 
> 
"roslaunch my_dynamixel_workbench_tutorial velocity_controllers.launch"

I am using two MX-64 on a USB2dyanmixel. The pan&tilt tutorial worked fine. Traced the error to:

dynamixel_workbench_velocity_control.cpp:
> `bool DynamixelWorkbenchVelocityControl::writeVelocity(int64_t left_wheel_velocity, int64_t right_wheel_velocity)
> {
>   dynamixel_[LEFT_RIGHT_WHEEL]->item_ = dynamixel_[LEFT_RIGHT_WHEEL]->ctrl_table_["goal_velocity"];
> 
>   writeSyncDynamixel(dynamixel_[LEFT_RIGHT_WHEEL]->item_->address, dynamixel_[LEFT_RIGHT_WHEEL]->item_->data_length, left_wheel_velocity, right_wheel_velocity);
> }`

I do not believe DXL 1.0 can use "goal_velocity". When I changed that to "moving_speed" and butchered convertVelocity2Value to look like:

> int64_t DynamixelWorkbenchVelocityControl::convertVelocity2Value(double velocity)
> {
>   if(velocity>200)
>     velocity = 200;
>   if(velocity<0)
>     velocity = 0;
>   
>   velocity=velocity*(2047/200.0);
>   return (int64_t) (velocity);
> 
> //  return (int64_t) (velocity * dynamixel_[LEFT_RIGHT_WHEEL]->velocity_to_value_ratio_);
> }

everything worked. Though not as you intend in the tutorial. Input [0-200] for each velocity value where 0 to 100 goes one way and 101 to 200 goes the other way. Not sure how you fix this as I don't have any DXL 2.0 motors.

>

routiful, 2016-10-21:
Hi, airduke
Sorry for my late reply.
You've tried in right way, I appericiated your comment. 
I uploaded modified source on the develop branch.
Please try and let me know if it works fine. 
Thank you.

airduke, 2016-10-25:
Works fine! 
I know its just a basic tutorial but it would be helpful to add another button besides wasd that stops all motors. Right now you have to undo everything you've done or close the velocity controller. This would make the tutorial a better fallback for other testing.

Thanks

closed 2016-10-25 by airduke

routiful, 2016-10-26:
Hello, airduke
I agree with your comments. The next release will be added stop signal in velocity controller.
Thank you for your feedbacks.