ROBOTIS-GIT/dynamixel-workbench #27: meaning of max_torque, torque_enable, torque_limit?

flobotics, 2017-01-07:
hi,
i got  AX12-a servos, but when i read the "load" value from dynamixel_msg/JointState message, it only tells me value around 0.0 and 0.02 or -0.01 ? When i put load on the servo it should increase to 1023, or am i wrong ?

Do i need to enable the torque? I use the servos in wheel-mode

What are the differences or max_torque,torque_limit and torque_enable ?

thanks

routiful, 2017-01-11:
Hi, flobotics :)

When I checked 'dynamixel_workbench_single_manager' with AX-12A (wheel-mode), a value of the 'present load' was increased '0 - 1023' (CCW direction) and '1024 - 2047' (CW direction) even torque_enable value was '0' . I think you need to update firmware of AX-12A or factory reset of it.

A 'max_torque' and a 'torque_limit' are related to PWM. The differences of them is address location. The address of the 'max_torque' is located in EEPROM and the address of the 'torque_limit' is saved in RAM. So, 'max_torque' data will not deleted after power off. But 'torque_limit' data is going to set initial value after power on. 

A 'torque_enable' is to generates torque by impressing the power to the power. So you have to set '1' by this address, if you want to operate your Dynamixel.

If you need more details about Dynamixel series, you can search information on [e-manual ](http://support.robotis.com/en/) or contact to _support@robotis.com_

Thank you.

flobotics, 2017-01-11:
hi routiful,
thanks for your answer. And i see a increasing value like you said in "present_load". Why is this value not present in dynamixel/JointState   http://docs.ros.org/kinetic/api/dynamixel_msgs/html/msg/JointState.html ?

How can i get this value with a ros topic ?

Another question is, the torque goes away and does not stall after moving ? If i try to move the servo, it should stall and build up torque to hold it and not move backwards until the torque is zero, or ?

routiful, 2017-01-12:
Hi, flobotics :>

[dynamixel_controllers](http://wiki.ros.org/dynamixel_controllers)' packages are not official [ROBOTIS](http://wiki.ros.org/ROBOTIS). So, I don't know how to calculate a 'load' value and what value is published. Maybe, you are going to be serviced by maintainers related 'dynamixel_controllers'  after you make issues on their github pages about this. 

Additionally, a 'present_load' published by 'dynamixel_workbench_single_manager' is raw value which is read in AX-12A and this motor provides position and velocity control mode. If you want to use current_control (similar definition with torque_control), you need to buy Dynamixel-X series (XM350-W210-R or XM350-W350-R).

Thank you

closed 2017-02-02 by routiful