ROBOTIS-GIT/dynamixel-workbench #32: Fail to launch single manager node

alanypf, 2017-04-12:
Hi there,

I am using AX-12A and the parameters for the single node manager launch file are:
arg name="device_name" default="/dev/ttyACM0"/
arg name="baud_rate" default="1000000"
The result is:
[ INFO] [1491987416.102719320]: Succeeded to open the port(/dev/ttyACM0)!
[ INFO] [1491987416.103710709]: Succeeded to change the baudrate(1000000)!
[ INFO] [1491987416.103746418]: Scan Dynamixel Using Protocol 1.0
[ERROR] [1491987416.532202892]: Unable to open model_info file : /opt/ros/kinetic/share/dynamixel_workbench_toolbox/dynamixel/models/model_info.list

I have no idea what went wrong, any suggestions?
Thank you!

routiful, 2017-04-12:
Hi :)

You can get latest code in master branch.
Before clone it your catkin_workspace, you have to purge last code you installed.

Thanks

alanypf, 2017-04-13:
It is all good now.
Thanks.

closed 2017-04-13 by alanypf