ROBOTIS-GIT/dynamixel-workbench #34: dynamixel MX-28AT can not open when run the dynamixel workbench single manager  tutorials

wilsionhuang, 2017-04-13:
Dear all
     I'm sorry to disturb us all.i just confuse on running the Dynamixel MX-28AT when i try to run the roslaunch dynamixel_workbench_single_manager dynamixel_workbench_single_manager.launch.
and i get the error:
 INFO] [1492075464.593187177]: dynamixel_workbench_single_manager : Init OK!
[ INFO] [1492075464.593977160]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1492075464.594724793]: Succeeded to change the baudrate(57600)!
[ INFO] [1492075464.594750092]: Scan Dynamixel Using Protocol 1.0
[ INFO] [1492075467.378554291]: Scan Dynamixel Using Protocol 2.0
[ERROR] [1492075467.431782271]: Unable to open .device file : /home/wilsion/catkin_ws/src/dynamixel-workbench-master/dynamixel_workbench_toolbox/dynamixel/models//.device
================================================================================REQUIRED process [dynamixel_workbench_single_manager-2] has died!
process has died [pid 9334, exit code -11, cmd /home/wilsion/catkin_ws/devel/lib/dynamixel_workbench_single_manager/dynamixel_workbench_single_manager __name:=dynamixel_workbench_single_manager __log:=/home/wilsion/.ros/log/fb9ad808-202a-11e7-a3e3-34f39a3a1b1e/dynamixel_workbench_single_manager-2.log].
log file: /home/wilsion/.ros/log/fb9ad808-202a-11e7-a3e3-34f39a3a1b1e/dynamixel_workbench_single_manager-2*.log
Initiating shutdown!
================================================================================
[dynamixel_workbench_single_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

when i try to search the device file : /home/wilsion/catkin_ws/src/dynamixel-workbench-master/dynamixel_workbench_toolbox/dynamixel/models//.device .i found that in the MX_28.device,but i can not found  error .
and when i change the dynamixel MX-28.it works well.so can anyone help me or give me some advice?
thanks!!

routiful, 2017-04-13:
Hi :)

[Issue 32](https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/32) may help you.

Thanks.

wilsionhuang, 2017-04-13:
sorry,although i use the master branch to run,it's also appear the same error when i try to control the dynamixel MX-28AT.But if i change the MX-28 ,it works well.so i confused if the device file of MX_28.device is not consuitable to MX-28AT.
note:MX-28 MX-28AT is two different type motor.
so can you give me some advice?

routiful, 2017-04-14:
Hi again :)

I try to control MX-28AT using dynamixel-workbench. It works fine.
MX-28AT and MX-28 are different cover but same firmware. So, dynamixel-workbench toolbox has only MX-28. You may update firmware MX-28AT using roboplus and try again single_manager.

Thanks.

wilsionhuang, 2017-04-14:
Hi,sorry to disturb you again,

First,thank you for your advice,but i have tried to update the firmware on roboplus 2.0,it also donn't work while it can be run in the roboplus on windows when i update the fireware,so it confuses me.i do not know which step of updating firmware may ignored .i think the device can not be found is the major problem. i'm glad to receive your reply.
Thanks!

routiful, 2017-04-14:
Hi ;)

MX-28AT which is used protocol 1.0 would be connected with [ROBOPLUS 1.0](http://en.robotis.com/BlueAD/board.php?bbs_id=downloads&mode=view&bbs_no=1152561&page=1&key=&keyword=&sort=&scate=)
This program supports firmware upgrade to MX series. 
If you have any problem after firmware upgrade, notice me freely.

Thanks.

closed 2017-04-23 by routiful