ROBOTIS-GIT/dynamixel-workbench #40: unable to read the dynamixel model MX_28

Zero23, 2017-05-02:
following is the terminal message showing:
PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /velocity_controller_tutorials/baud_rate: 57142
 * /velocity_controller_tutorials/device_name: /dev/ttyUSB0
 * /velocity_controller_tutorials/left_motor/motor_id: 1
 * /velocity_controller_tutorials/motor_model: MX_28
 * /velocity_controller_tutorials/protocol_version: 2.0
 * /velocity_controller_tutorials/right_motor/motor_id: 2

NODES
  /
    velocity_controller_tutorials (dynamixel_workbench_controllers/dynamixel_workbench_velocity_control)

auto-starting new master
process[master]: started with pid [3015]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d439a2ba-2ee6-11e7-a59a-fcaa14dae030
WARNING: Package name "speedTester" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[rosout-1]: started with pid [3028]
started core service [/rosout]
process[velocity_controller_tutorials-2]: started with pid [3035]
[ INFO] [1493695460.178438429]: dynamixel_workbench_velocity_control : Init OK!
[ INFO] [1493695460.180275489]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1493695460.180742480]: Succeeded to change the baudrate(57142)!
[ INFO] [1493695460.181342162]: left_motor_id: 1
[ INFO] [1493695460.181387583]: left_motor_model: MX_28
[ INFO] [1493695460.181420362]: left_motor_protocol_version_: 2.0

[ INFO] [1493695460.224465497]: right_motor_id: 2
[ INFO] [1493695460.224511827]: right_motor_model: MX_28
[ INFO] [1493695460.224549510]: right_motor_protocol_version_: 2.0
[ERROR] [1493695460.237113647]: [ID] 1, Fail to read!
[ERROR] [1493695460.249320314]: [ID] 2, Fail to read!
[ERROR] [1493695460.261483263]: [ID] 1, Fail to read!
[ERROR] [1493695460.273646766]: [ID] 2, Fail to read!
[ERROR] [1493695460.285988967]: [ID] 1, Fail to read!

routiful, 2017-05-08:
Hi :)

dynamixel_workbench supports MX-28 has protocol version 1.0 only. 
MX-28 has protocol version 2.0 will be supported soon.
So, change a protocol version using single_manager first and launch velocity controller again.

Thanks.

closed 2017-05-24 by routiful

ct0227, 2017-11-06:
Hi, I also have this kind of question and it shows like this:

PARAMETERS
 * /axis_angular: 0
 * /axis_linear: 1
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /scale_angular: 0.5
 * /scale_linear: 3.0
 * /twist_mux/locks: [{'topic': 'e_sto...
 * /twist_mux/topics: [{'topic': 'joy_t...
 * /use_gui: True
 * /velocity_controller_tutorials/baud_rate: 1000000
 * /velocity_controller_tutorials/center_motor/motor_id: 3
 * /velocity_controller_tutorials/device_name: /dev/ttyUSB0
 * /velocity_controller_tutorials/left_motor/motor_id: 1
 * /velocity_controller_tutorials/motor_model: MX_64
 * /velocity_controller_tutorials/protocol_version: 1.0
 * /velocity_controller_tutorials/right_motor/motor_id: 2

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)
    rviz (rviz/rviz)
    twist_mux (twist_mux/twist_mux)
    velocity_controller_tutorials (dynamixel_workbench_controllers/dynamixel_workbench_velocity_control)
    workbench_keyboard (dynamixel_workbench_tutorials/dynamixel_workbench_velocity_conversion)

auto-starting new master
process[master]: started with pid [10247]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a75f3a84-c2ca-11e7-a2a9-801934e71733
process[rosout-1]: started with pid [10260]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [10277]
process[robot_state_publisher-3]: started with pid [10278]
process[rviz-4]: started with pid [10279]
process[velocity_controller_tutorials-5]: started with pid [10280]
process[workbench_keyboard-6]: started with pid [10282]
[ INFO] [1509956131.528811450]: dynamixel_workbench_velocity_control : Init OK!
process[twist_mux-7]: started with pid [10309]
[ INFO] [1509956131.539445420]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1509956131.539951071]: Succeeded to change the baudrate(1000000)!
[ INFO] [1509956131.541495262]: left_motor_id: 1
[ INFO] [1509956131.541540980]: left_motor_model: MX_64
[ INFO] [1509956131.541563410]: left_motor_protocol_version_: 1.0

[ INFO] [1509956131.573490415]: right_motor_id: 2
[ INFO] [1509956131.573727945]: right_motor_model: MX_64
[ INFO] [1509956131.573893491]: right_motor_protocol_version_: 1.0

[ INFO] [1509956131.669761252]: center_motor_id: 3
[ INFO] [1509956131.669877231]: center_motor_model: MX_64
[ INFO] [1509956131.670004929]: center_motor_protocol_version_: 1.0
[ERROR] [1509956131.681138072]: [ID] 1, Fail to read!
[ERROR] [1509956131.693072987]: [ID] 3, Fail to read!
[ERROR] [1509956131.709079037]: [ID] 2, Fail to read!
[ERROR] [1509956131.725097686]: [ID] 1, Fail to read!
[ERROR] [1509956131.741097758]: [ID] 3, Fail to read!
[ERROR] [1509956131.757052049]: [ID] 2, Fail to read!
[ERROR] [1509956131.773045657]: [ID] 1, Fail to read!
[ERROR] [1509956131.789032408]: [ID] 3, Fail to read!
[ERROR] [1509956131.805101652]: [ID] 2, Fail to read!
[ERROR] [1509956131.821121297]: [ID] 1, Fail to read!
[ERROR] [1509956131.837040301]: [ID] 3, Fail to read!
[ERROR] [1509956131.853827017]: [ID] 2, Fail to read!
[ERROR] [1509956131.869006103]: [ID] 1, Fail to read!
[ERROR] [1509956131.884997934]: [ID] 3, Fail to read!
[workbench_keyboard-6] process has finished cleanly
log file: /home/atsushi/.ros/log/a75f3a84-c2ca-11e7-a2a9-801934e71733/workbench_keyboard-6*.log
[ERROR] [1509956131.904988972]: [ID] 2, Fail to read!
[ERROR] [1509956131.917188258]: [ID] 1, Fail to read!
[ERROR] [1509956131.932917538]: [ID] 3, Fail to read!
[ERROR] [1509956131.948920482]: [ID] 2, Fail to read!
[ERROR] [1509956131.964927875]: [ID] 1, Fail to read!
[ERROR] [1509956131.980909218]: [ID] 3, Fail to read!
[ERROR] [1509956131.996899554]: [ID] 2, Fail to read!
[ERROR] [1509956132.013141471]: [ID] 1, Fail to read!
[ERROR] [1509956132.029112547]: [ID] 3, Fail to read!
[ERROR] [1509956132.044874639]: [ID] 2, Fail to read!
[ERROR] [1509956132.060831617]: [ID] 1, Fail to read!
[ERROR] [1509956132.076860287]: [ID] 3, Fail to read!
[ERROR] [1509956132.092855748]: [ID] 2, Fail to read!
[ERROR] [1509956132.108892658]: [ID] 1, Fail to read!
[ERROR] [1509956132.124813118]: [ID] 3, Fail to read!
[ERROR] [1509956132.141651345]: [ID] 2, Fail to read!

And I already check the id and baudrate which are all correct. Does anyone know why? Thanks a lot!!

routiful, 2017-11-07:
Hi @ct0227 

[OpenManipulator](https://github.com/ROBOTIS-GIT/open_manipulator) uses dynamixel workbench to control 5 dynamixel.
This example will help you.

Thanks
Darby