ROBOTIS-GIT/dynamixel-workbench #45: Fail to read!

wilsionhuang, 2017-05-24:
when i try to use the dynamixel_workbench_position_control read dynamixel pro data, it appear error:[ERROR] [1495609283.918645613]: [ID] 5, Fail to read.and i have check the data         dynamixel_comm_result_ = packetHandler_->read4ByteTxRx(portHandler_, id, addr, (uint32_t*)&value_32_bit, &dynamixel_error);but the motor id is right,so i have no idea of how to solve it.
can anyone give some advice to me ! 
thanks advance!

routiful, 2017-05-25:
Hi ;)

Did you use 24V power for operating DYNAMIXEL Pro?
If you supply lower voltage to DYNAMIXEL Pro, it is not working properly.
Please, check it first.

Thanks

wilsionhuang, 2017-05-25:
Hi:
Firstly,thanks for your response,I'm sure that i use the 24V power for operating DYNAMIXEL PRO ,i have some  trouble of the read4ByteTxRx(),and  the data of  dynamixel_comm_result_ is 71 so confused ! Thus the loop end by failed to read.i also check the parameters, id is right,so i do not know where is wrong.

routiful, 2017-06-12:
Hi :)

I am checking your problem. It will take one or two days. 
After finish it, i notice you.
sorry for inconvenient.

Thanks.

routiful, 2017-06-23:
Hello :)

I uploaded new version of dynamixel-workbench.
I checked Dynamixel PRO is working fine in new version.
the new version has been changed some parameters. 
So you need to check [tutorial](http://wiki.ros.org/dynamixel_workbench/Tutorials) first

Thanks.

closed 2017-07-04 by routiful