ROBOTIS-GIT/dynamixel-workbench #47: ROS Indigo support

roboshiba, 2017-05-31:
Sorry to be a very rudimentary question.
Does Dynamixel_Workbench support Indigo?
On the ROS wiki the status is "Can not load information".

Some ROS user communities in Japan unify ROS version in Indigo to share resources. (Especially rescue robot researcher community)

Multiple questions came to ROBOTIS Japan. I will ask as representative.

Thanks

routiful, 2017-06-07:
Hi :)

dynamixel_workbench is compatible with ROS indigo version.
So, users who use ROS indigo can clone dynamixel_workbench package to their catkin workspace. I tested it two month ago.

If errors occur in ROS indigo, please feel free to open issue.

Thanks

roboshiba, 2017-06-08:
I am pleased that your official view was obtained.
Thanks.

closed 2017-06-08 by roboshiba