ROBOTIS-GIT/dynamixel-workbench #48: cannot install on Raspberry Pi

ryanoldja, 2017-06-11:
Running `sudo-apt-get install ros-kinetic-dynamixel-workbench ` works for me on my Ubuntu 16.04 but fails on my Raspberry Pi Jessie with `E: unable to locate package ros-kinetic-dynamixel-workbench`. None of its dependent packages can be installed in the same way either.

Can any sources be added to /etc/apt/sources.list or can the package be installed in another way?

routiful, 2017-06-12:
HI :)

I didn't install dynamixel_workbench on Raspberry Pi.
Have you tried cloning dynamixel_workbench in catkin workspace?
I would appreciate it if you let me know how it works on RPi.

Thanks

ryanoldja, 2017-06-13:
Hi,

I created a new catkin workspace and cloned dynamixel-workbench and DynamixelSDK packages into src.
The following is the terminal output upon trying to build the workspace:
`
pi@raspberrypi:~/dynamixel_workbench_ws $ catkin_make
Base path: /home/pi/dynamixel_workbench_ws
Source space: /home/pi/dynamixel_workbench_ws/src
Build space: /home/pi/dynamixel_workbench_ws/build
Devel space: /home/pi/dynamixel_workbench_ws/devel
Install space: /home/pi/dynamixel_workbench_ws/install
####
#### Running command: "cmake /home/pi/dynamixel_workbench_ws/src -DCATKIN_DEVEL_PREFIX=/home/pi/dynamixel_workbench_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pi/dynamixel_workbench_ws/install -G Unix Makefiles" in "/home/pi/dynamixel_workbench_ws/build"
####
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/pi/dynamixel_workbench_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.9") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/pi/dynamixel_workbench_ws/build/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - dynamixel_workbench (metapackage)
-- ~~  - dynamixel_workbench_msgs
-- ~~  - dynamixel_sdk
-- ~~  - dynamixel_workbench_toolbox
-- ~~  - dynamixel_workbench_controllers
-- ~~  - dynamixel_workbench_single_manager
-- ~~  - dynamixel_workbench_single_manager_gui
-- ~~  - dynamixel_workbench_tutorials
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'dynamixel_workbench'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench)
-- +++ processing catkin package: 'dynamixel_workbench_msgs'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- dynamixel_workbench_msgs: 16 messages, 3 services
-- +++ processing catkin package: 'dynamixel_sdk'
-- ==> add_subdirectory(DynamixelSDK/ros)
-- +++ processing catkin package: 'dynamixel_workbench_toolbox'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_toolbox)
-- +++ processing catkin package: 'dynamixel_workbench_controllers'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_controllers)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'dynamixel_workbench_single_manager'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_single_manager)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'dynamixel_workbench_single_manager_gui'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_single_manager_gui)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "qt_build" with any
  of the following names:

    qt_buildConfig.cmake
    qt_build-config.cmake

  Add the installation prefix of "qt_build" to CMAKE_PREFIX_PATH or set
  "qt_build_DIR" to a directory containing one of the above files.  If
  "qt_build" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_single_manager_gui/CMakeLists.txt:10 (find_package)

-- Could not find the required component 'qt_build'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "qt_build" with any
  of the following names:

    qt_buildConfig.cmake
    qt_build-config.cmake

  Add the installation prefix of "qt_build" to CMAKE_PREFIX_PATH or set
  "qt_build_DIR" to a directory containing one of the above files.  If
  "qt_build" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_single_manager_gui/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/dynamixel_workbench_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/dynamixel_workbench_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed`

routiful, 2017-06-13:
Hi :)

You should install "qt_build"

$ sudo apt-get install ros-{$ros-distro}-qt-build

Thanks

ryanoldja, 2017-06-13:
Sorry, didn't update my original post in time. I was unable to install qt_build through the apt-get repo.

`
pi@raspberrypi:~/dynamixel_motor_ws $ sudo apt-get install ros-kinetic-qt-build
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-qt-build`

routiful, 2017-06-13:
Did you try update and upgrade??

$ sudo apt-get update
$ sudo apt-get upgrade

ryanoldja, 2017-06-14:
Yep, ran
`sudo apt-get update`
`sudo apt-get upgrade`

still unable to install through apt-get

routiful, 2017-06-14:
I have tried update&upgrade and intall qt-build RPi3 with Ubuntu mate. It will be fine.

But dynamixel_workbench_single_manager_gui is not working on RPi3 even has required installation.
You can use dynamixel_workbench_single_manager in command line on it after delete dynamixel_workbench_single_manager_gui package.

Thanks.

ryanoldja, 2017-06-14:
We are running Raspbian Jessie, not MATE. So there seems to be an incompatibility issue with Raspbian. It would be nice to not shift to MATE at this point, if possible.

routiful, 2017-06-15:
I didn't try in Raspbian, due to dynamixel_workbench is developed considering Ubuntu environments.
But, I guess ,except dynamixel_workbench_single_manager_gui package, dynamxiel_workbench can be worked in it.

Thanks for sharing your circumstance.

closed 2017-06-23 by routiful