ROBOTIS-GIT/dynamixel-workbench #49: cannot launch node of type [dynamixel_workbench_single_manager]

jugal94, 2017-06-15:
Hi,

I am working with XH430-W350. on launching dynamixel_workbench_single_manager.launch I get the following error, 

ERROR: cannot launch node of type [dynamixel_workbench_single_manager/dynamixel_workbench_single_manager]: dynamixel_workbench_single_manager
ROS path [0]=/opt/ros/jade/share/ros
ROS path [1]=/opt/ros/jade/share
ROS path [2]=/opt/ros/jade/stacks

I also tried building the package again using catkin_make, which gave me the following error

CMake Error at /opt/ros/jade/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "dynamixel_sdk"
  with any of the following names:

    dynamixel_sdkConfig.cmake
    dynamixel_sdk-config.cmake

  Add the installation prefix of "dynamixel_sdk" to CMAKE_PREFIX_PATH or set
  "dynamixel_sdk_DIR" to a directory containing one of the above files.  If
  "dynamixel_sdk" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  dynamixel-workbench-master/dynamixel_workbench_controllers/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/developer-admin/dynamixel_jugal_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/developer-admin/dynamixel_jugal_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

routiful, 2017-06-16:
Hi :)

'dynamixel_workbench' depends on [dynamixel_sdk](https://github.com/ROBOTIS-GIT/DynamixelSDK)
You should install dynamixel_sdk

$ sudo apt-get install ros-{$ros_distro}-dynamixel-sdk*

I didn't work on Jade.
I'll be thankful if you let me know if it works.

Thanks.

jugal94, 2017-06-22:
Hi,

I upgraded my pos version to kinetic. I performed the sudo apt-get like you advised. I am able to run all the tutorials with roslaunch commands, the single_gui manager.

But, rostopic echo /velocity_control/dynamixel_state displays topic [/velocity_control/dynamixel_state] does not appear to be published

also, rosrun dynamixel_workbench_operators wheel_operator returns 
[rospack] Error: package 'dynamical_workbench_operators' not found

rosservice call /wheel_command 10 20 returns
Error: service [/wheel_command] is not available

Thank you for your help, I would like to know what I am doing wrong.

routiful, 2017-06-23:
Hi :)

dynamixel-workbench new version is uploaded.
you can check it [wiki](http://wiki.ros.org/dynamixel_workbench)

sorry for inconvenience,.

Thanks.

jugal94, 2017-06-23:
I tried the git clone method in the wiki page.

Yes, it does have the dynamixel_operator but invoking cmake failed this is the error that I faced, I checked the file and I saw that dynamixel_workbench_msgs is missing.

/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_operators/src/joint_operator.cpp:34:51: fatal error: dynamixel_workbench_msgs/JointCommand.h: No such file or directory
/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_operators/src/wheel_operator.cpp:37:51: fatal error: dynamixel_workbench_msgs/WheelCommand.h: No such file or directory
compilation terminated.
compilation terminated.
dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/src/wheel_operator.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/src/wheel_operator.cpp.o] Error 1
dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/src/joint_operator.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/src/joint_operator.cpp.o] Error 1
CMakeFiles/Makefile2:931: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1115: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/all] Error 2
[ 16%] Building CXX object dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_multi_driver.cpp.o
/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:61:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, uint16_t model_number)
 ^
In file included from /home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:33:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:74:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, std::string model_name)
 ^
In file included from /home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:33:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp: In member function ‘bool dynamixel_tool::DynamixelTool::getModelName(uint16_t)’:
/home/developer/dynamixel_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:91:3: error: ‘name_path_’ was not declared in this scope
   name_path_  = ros::package::getPath("dynamixel_workbench_toolbox") + "/dynamixel/model_info.list";
   ^
dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1184: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

I appreciate all your help till now. Thank you.

routiful, 2017-06-26:
Hi :)

Sorry, I didn't notice that [dynamixel_workbench_msgs](https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git) packages are separated. You need to download new dynamixel_workbench_msgs. It was changed a little bit last version. I add dependent package section in [WIKI](http://wiki.ros.org/dynamixel_workbench)

And you have to delete last version of dynamixel_workbench
this command will help you

$ sudo apt-get purge ros-kinetic-dynamixel-workbench*

Thanks

jugal94, 2017-06-30:
Thank you. It works fine with the update now.

I have two issues

1. I have 6 dynamixel motors that I want to work together. 4 XH series and 2 Pro L42. Is it possible to modify your code to run them together ?

2. The Pro motors would not run from single_manager as well as controllers. The package would fail ro recognise after scanning protocol 1.0 and protocol 2.0

Thank you

routiful, 2017-07-02:
Hi :)

1. 6 dynamixel motors can control to use dynamixel-workbench controller. The tutorial is [here](http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials).

2. I tested Pro motors with new version of dynamixel-workbench. The Pro motors needs 24V input voltage. Please check it and show me which error was occurred.

Thanks

jugal94, 2017-07-03:
Thank you for your reply.

1. Since there is no specific tutorial for connecting more than 2 motors with a single USB2Dynamixel, I am assuming that making changes to the launch file shown in the position control launch file would help run the 4 X series motor together that I have purchased. Please let me know if my assumption is wrong,

I tried using position controller tutorial shown on the wiki page, but i am facing some difficulties. 

**1.a** Launching the file displays the following error

ERROR: cannot launch node of type [dynamixel_workbench_controllers/position_control]: can't locate node [position_control] in package [dynamixel_workbench_controllers]

To which i referred the code for dynamixel_workbench_position_controller in the package dynamixel_workbench_controller, i found that the following lines are different

--> from tutorial
 <node name="potision_control_example" pkg="dynamixel_workbench_controllers" type="position_control" required="true" output="screen">

--> from dynamixel_workbench package
 <node name="potision_control_example" pkg="dynamixel_workbench_controllers" type="dynamixel_workbench_position_control" required="true" output="screen">

can you confirm that the line should have type = dynamixel_workbench _position_control and not type = position_control as shown in the tutorial.

As, I saw the executable with name dynamixel_workbench _position_control being created in the lib space after using sudo apt-get ...

Also running the launch file with the above mentioned change to the type = .. gave me the following error:

SUMMARY
========

PARAMETERS
 * /baud_rate: 57600
 * /device_name: /dev/ttyUSB0
 * /pan_id: 3
 * /potision_control_example/profile_acceleration: 10
 * /potision_control_example/profile_velocity: 100
 * /protocol_version: 2.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /tilt_id: 4

NODES
  /
    potision_control_example (dynamixel_workbench_controllers/dynamixel_workbench_position_control)

auto-starting new master
process[master]: started with pid [20818]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 32dcb5ae-6007-11e7-a36b-0024e83bdd55
process[rosout-1]: started with pid [20831]
started core service [/rosout]
process[potision_control_example-2]: started with pid [20845]
[PortHandlerLinux::SetupPort] Error opening serial port!
[ERROR] [1499096969.904401109]: Failed to open the port!
[PortHandlerLinux::SetupPort] Error opening serial port!
[PortHandlerLinux::SetCustomBaudrate] TIOCGSERIAL failed!
[ERROR] [1499096969.904498154]: Failed to change the baudrate!
[ INFO] [1499096969.905221285]: pan_motor_id: 0
[ INFO] [1499096969.905259552]: pan_motor_model: 
[ INFO] [1499096969.905287197]: pan_motor_protocol_version_: 0.0

[ERROR] [1499096969.919153334]: Unable to open .device file : /opt/ros/kinetic/share/dynamixel_workbench_toolbox/dynamixel/models//.device
================================================================================REQUIRED process [potision_control_example-2] has died!
process has died [pid 20845, exit code -11, cmd /opt/ros/kinetic/lib/dynamixel_workbench_controllers/dynamixel_workbench_position_control __name:=potision_control_example __log:=/home/developer/.ros/log/32dcb5ae-6007-11e7-a36b-0024e83bdd55/potision_control_example-2.log].
log file: /home/developer/.ros/log/32dcb5ae-6007-11e7-a36b-0024e83bdd55/potision_control_example-2*.log
Initiating shutdown!
================================================================================
[potision_control_example-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Would love your assistance on this

2. The following error is displayed for Pro L42 motors on single manager

process[dynamixel_workbench_single_manager-2]: started with pid [21329]
[ INFO] [1499098879.192649565]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1499098879.202544683]: Succeeded to change the baudrate(57600)!
[ INFO] [1499098879.202596212]: Scan Dynamixel Using Protocol 1.0
[ INFO] [1499098884.003804653]: Scan Dynamixel Using Protocol 2.0
[ INFO] [1499098889.048001661]: ...Succeeded to find dynamixel

[ INFO] [1499098889.048060162]: [ID] 1, [Model Name] PRO_L42_10_S300_R, [BAUD RATE] 57600
[ INFO] [1499098889.054041595]: ----------------------------------------------------------------------
[ INFO] [1499098889.054081071]: Single Manager supports GUI (dynamixel_workbench_single_manager_gui)  
[ INFO] [1499098889.054104581]: ----------------------------------------------------------------------
[ INFO] [1499098889.054134552]: Command list :
[ INFO] [1499098889.054153887]: [help|h|?]................: display this menu
[ INFO] [1499098889.054174624]: [status]..................: status of a Dynamixel
[ INFO] [1499098889.054194886]: [table]...................: check a control table of a dynamixel
[ INFO] [1499098889.054214894]: [reboot]..................: reboot a Dynamixel(only protocol version 2.0)
[ INFO] [1499098889.054237598]: [factory_reset]...........: command for all data back to the factory settings values
[ INFO] [1499098889.054260772]: [[table_item] [value].....: change address value of a dynamixel
[ INFO] [1499098889.054282065]: [exit]....................: shutdown
[ INFO] [1499098889.054310643]: ----------------------------------------------------------------------
[ INFO] [1499098889.054333010]: Press SpaceBar to command a Dynamixel
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!

and then the last line goes on repeating

Thank you for all your assistance till now.

routiful, 2017-07-04:
Hi :)

The error messages show last version of dynamixel-workbench.
You can use this command to delete it.

$ sudo apt-get purge ros-kinetic-dynamixel-workbench*

After delete installed metapackage, clone(master branch) it your catkin_workspace.

Thanks.

jugal94, 2017-07-05:
I cloned the master package

I could not evoke catkin_make successfully and got the following error. I saw this error before and you solved it for me as well. but now it is coming back up again.

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "dynamixel_workbench_msgs" with any of the following names:

    dynamixel_workbench_msgsConfig.cmake
    dynamixel_workbench_msgs-config.cmake

  Add the installation prefix of "dynamixel_workbench_msgs" to
  CMAKE_PREFIX_PATH or set "dynamixel_workbench_msgs_DIR" to a directory
  containing one of the above files.  If "dynamixel_workbench_msgs" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_operators/CMakeLists.txt:12 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "dynamixel_workbench_msgs" with any of the following names:

    dynamixel_workbench_msgsConfig.cmake
    dynamixel_workbench_msgs-config.cmake

  Add the installation prefix of "dynamixel_workbench_msgs" to
  CMAKE_PREFIX_PATH or set "dynamixel_workbench_msgs_DIR" to a directory
  containing one of the above files.  If "dynamixel_workbench_msgs" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_operators/CMakeLists.txt:12 (find_package)

routiful, 2017-07-06:
Hi :)

You need to clone [dynamixel-workbench-msgs](https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git). 
Before you catkin_make, delete 'devel' and 'build' folder in catkin_workspace.

Thanks.

routiful, 2017-07-07:
Hi :)

I made a [package](https://github.com/ROBOTIS-GIT/open_manipulator) to control four dynamixel using dynamixel-workbench-toolbox
It may help your work.

Thanks.

jugal94, 2017-07-10:
Two motors connected together work fine now. Thank you for all the help. This package seems to be super helpful. I will try it out. 

I am closing this issue and will open issues on the new package like if any.

closed 2017-07-10 by jugal94

Monchou2016, 2018-11-21:
Hello,

I saw this thread. I am not sure if my problem is related to this thread but I got these errors below. I'm new to ROS btw. Thank you in advance for the help.

/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘virtual void DYNAMIXEL::JointDynamixel::enable()’:
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:70:36: error: ‘class DynamixelWorkbench’ has no member named ‘torqueOn’
     result = dynamixel_workbench_->torqueOn(dynamixel_.id.at(index), &
                                    ^
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘virtual void DYNAMIXEL::JointDynamixel::disable()’:
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:85:36: error: ‘class DynamixelWorkbench’ has no member named ‘torqueOff’
     result = dynamixel_workbench_->torqueOff(dynamixel_.id.at(index), 
                                    ^
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::initialize(std::vector<unsigned char>, STRING, STRING)’:
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:127:34: error: ‘class DynamixelWorkbench’ has no member named ‘init’
   result = dynamixel_workbench_->init(dxl_device_name.c_str(), std::at
                                  ^
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:137:68: error: no matching function for call to ‘DynamixelWorkbench::ping(uint8_t&, uint16_t*, const char**)’
     result = dynamixel_workbench_->ping(id, &get_model_number, &log);
                                                                    ^
In file included from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:49:8: note: candidate: bool DynamixelWorkbench::ping(uint8_t, uint16_t*)
   bool ping(uint8_t id, uint16_t *get_model_number = 0);
        ^
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:49:8: note:   candidate expects 2 arguments, 3 provided
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::setOperatingMode(std::vector<unsigned char>, STRING)’:
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:168:91: error: no matching function for call to ‘DynamixelWorkbench::jointMode(__gnu_cxx::__alloc_traits<std::allocator<unsigned char> >::value_type&, const uint32_t&, const uint32_t&, const char**)’
 el_workbench_->jointMode(actuator_id.at(num), velocity, effort, &log);
                                                                     ^
In file included from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note: candidate: bool DynamixelWorkbench::jointMode(uint8_t, uint16_t, uint16_t)
   bool jointMode(uint8_t id, uint16_t vel = 0, uint16_t acc = 0);
        ^
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note:   candidate expects 3 arguments, 4 provided
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:179:38: error: ‘class DynamixelWorkbench’ has no member named ‘CurrentBasedPositionMode’
       result = dynamixel_workbench_->CurrentBasedPositionMode(actuator
                                      ^
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:190:91: error: no matching function for call to ‘DynamixelWorkbench::jointMode(__gnu_cxx::__alloc_traits<std::allocator<unsigned char> >::value_type&, const uint32_t&, const uint32_t&, const char**)’
 el_workbench_->jointMode(actuator_id.at(num), velocity, effort, &log);
                                                                     ^
In file included from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note: candidate: bool DynamixelWorkbench::jointMode(uint8_t, uint16_t, uint16_t)
   bool jointMode(uint8_t id, uint16_t vel = 0, uint16_t acc = 0);
        ^
/home/monchou/catkin_ws/src/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note:   candidate expects 3 arguments, 4 provided
/home/monchou/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::setSDKHandler(uint8_t)’:
[... cut ...]
open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/build.make:110: recipe for target 'open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/src/Dynamixel.cpp.o' failed
make[2]: *** [open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/src/Dynamixel.cpp.o] Error 1
CMakeFiles/Makefile2:3612: recipe for target 'open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/all' failed
make[1]: *** [open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

routiful, 2018-11-22:
Hi :) you are supposed to download dynamixel_workbench_toolbox library in develop branch.

Monchou2016, 2018-11-22:
Thank you for your response. 
I cloned the dynamixel-workbench in catkin_ws/src. Under this folder, it contained many files including the dynamixel-workbench-toolbox. When trying I am to catkin_make, I still got such error.

routiful, 2018-11-27:
This issued might be help you #202