ROBOTIS-GIT/dynamixel-workbench #50: ROS -indigo error

elthef, 2017-06-16:
I encountered this error when I did catkin_make
/home/viki/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_controllers/src/dynamixel_workbench_multi_port.cpp:59:78: error: no matching function for call to ‘ros::NodeHandle::getParam(const char [27], float&)’
   nh_priv_.getParam("pan_motor/protocol_version", pan_motor_protocol_version_);
              
I am running nootriix 64bit in VM  VirtualBox.
What is the cause and ahowto rectify.
Thank you

routiful, 2017-06-16:
Hi :)

I will checking it.
Sorry for inconvenience.

Thanks.

routiful, 2017-06-23:
Hi :)

dynamixel-workbench new version is uploaded.
you can check it [wiki](http://wiki.ros.org/dynamixel_workbench)

sorry for inconvenience,.

Thanks.

elthef, 2017-06-23:
Thank you

Edwin

*********************************
Basic Information Before Leaving Earth<http://www.biblegateway.com/>

Foo Tuan Hoe Edwin

*********************************

closed 2017-06-28 by routiful