ROBOTIS-GIT/dynamixel-workbench #55: I have some questions about this workbench sources.

sh4j, 2017-06-28:
Hello,
Arbotix and previous source codes in ROBOTIS-GIT work well in the last week.

1. Why 'msgs' directory in 'devel branch' and its sources have been omitted in 'master branch'?
I have used ROS Indigo and catkin_make command worked well in the last week.

2. What is the different between kinetic-devel branch and devel branch now?
master branch has the problem about omitted 'msgs' directory so I can only use devel branch.

3. single_manager_gui cannot be complied with catkin_make in Indigo.
I have complied through catkin_make without single_manager_gui to complie completely.
There are errors like this:
`In file included from /home/kobuki/ca/src/git_dynamixel/dynamixel-workbench-kinetic-devel/dynamixel_workbench_single_manager_gui/src/qnode.cpp:38:0:
/home/kobuki/ca/src/git_dynamixel/dynamixel-workbench-kinetic-devel/dynamixel_workbench_single_manager_gui/src/../include/dynamixel_workbench_single_manager_gui/qnode.hpp:95:37: error: ‘std::__cxx11’ has not been declared
   void sendSetOperatingModeMsg(std::__cxx11::string index);
.... blah blah`

4.
Through the tutorial (http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials/MultiPort), 
my ROS cannot make and recognise any parameter written in multi_port.launch, args and params.
So I edited dynamixel_workbench_multi_port.cpp directly to recognise the dynamixels, it works.
But, it cannot make the model of the dynamixels and it attempts to access the directory
 [/home/kobuki/ca/src/git_dynamixel/dynamixel-workbench-kinetic-devel/dynamixel_workbench_toolbox/dynamixel/models//.device]
It seems ROS cannot take any args, names, params... as I think.
 It is most big problem for me.

How can I make the connection in ROS indigo to the dynamixels?

routiful, 2017-06-28:
Hi :)

1. 'msgs' package was separated. So you can find it [here](https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git). Updated version of dynamixel-workbench needs new 'dynamixel-workbench-msgs' package.

2. 'kinetic-devel' branch is used to upload for official ROS packages. It points that which version is downloaded by 'apt-get install' commands.

3. 'std::__cxx11' error is sometimes occurred. I added line for 'c++11' in cmake files. You can clone it master branch or delete it in source code. (std::__cxx11::string index -> std::string index)

4. Please let me know which dynamixel you used. Then, I'll test it.

Thanks

sh4j, 2017-06-28:
Thanks a lot!
I have installed from git (dynamixel-workbench-msgs) and it works well in Indigo!
I used msgs directory where is in the previous version.
Now, my MX_64 dynamixels are recognised well through the tutorial.

감사합니다 ^.^

closed 2017-06-28 by sh4j