ROBOTIS-GIT/dynamixel-workbench #58: Support for ROS Jade?

JoshMarino, 2017-07-10:
Hello,

Has anyone gotten this to work for ROS Jade? I tried cloning the packages into my work-space and then compiling it. I got an error saying that it cannot find qt_build required for dynamixel_workbench_single_manager_gui, which was not released for Jade.

Thanks

routiful, 2017-07-11:
Hi :)

It needs some dependent packages

$ sudo apt-get install ros-kinetic-dynamixel-sdk
$ sudo apt-get install ros-kinetic-qt-build

Pleas type above command on your terminal and build again.

Thanks.

JoshMarino, 2017-07-11:
I do not have access to the ros-kinetic packages since I am running Ubuntu 14.04. I tried using the ros-indigo versions and I was able to catkin_make the package now. When I try a catkin_make install through, it errors out.

-- Installing: /home/gecko/catkin_ws/install/lib/pkgconfig/dynamixel_sdk.pc
-- Up-to-date: /home/gecko/catkin_ws/install/share/dynamixel_sdk/cmake/dynamixel_sdkConfig.cmake
-- Up-to-date: /home/gecko/catkin_ws/install/share/dynamixel_sdk/cmake/dynamixel_sdkConfig-version.cmake
-- Up-to-date: /home/gecko/catkin_ws/install/share/dynamixel_sdk/package.xml
-- Up-to-date: /home/gecko/catkin_ws/install/lib/libdynamixel_sdk.so
-- Installing: /home/gecko/catkin_ws/install/include/dynamixel_sdk
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/group_bulk_read.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/group_bulk_write.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/protocol2_packet_handler.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/port_handler_windows.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/dynamixel_sdk.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/group_sync_write.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/group_sync_read.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/protocol1_packet_handler.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/port_handler_linux.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/packet_handler.h
-- Up-to-date: /home/gecko/catkin_ws/install/include/dynamixel_sdk/port_handler.h
-- Installing: /home/gecko/catkin_ws/install/lib/pkgconfig/dynamixel_workbench_operators.pc
-- Up-to-date: /home/gecko/catkin_ws/install/share/dynamixel_workbench_operators/cmake/dynamixel_workbench_operatorsConfig.cmake
-- Up-to-date: /home/gecko/catkin_ws/install/share/dynamixel_workbench_operators/cmake/dynamixel_workbench_operatorsConfig-version.cmake
-- Installing: /home/gecko/catkin_ws/install/share/dynamixel_workbench_operators/package.xml
-- Installing: /home/gecko/catkin_ws/install/lib/dynamixel_workbench_operators/joint_operator
-- Removed runtime path from "/home/gecko/catkin_ws/install/lib/dynamixel_workbench_operators/joint_operator"
-- Installing: /home/gecko/catkin_ws/install/lib/dynamixel_workbench_operators/wheel_operator
-- Removed runtime path from "/home/gecko/catkin_ws/install/lib/dynamixel_workbench_operators/wheel_operator"
CMake Error at dynamixel-workbench/dynamixel_workbench_operators/cmake_install.cmake:87 (FILE):
  file INSTALL cannot find
  "/home/gecko/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_operators/include/dynamixel_workbench_operators".
Call Stack (most recent call first):
  cmake_install.cmake:147 (INCLUDE)

make: *** [install] Error 1
Invoking "make install -j8 -l8" failed

JoshMarino, 2017-07-11:
Thanks, that worked, but I had to install qt_build for ROS Indigo since my machine is Ubuntu 14.04.

routiful, 2017-07-12:
qt_build does not support JADE??

closed 2017-08-02 by routiful