ROBOTIS-GIT/dynamixel-workbench #59: cmake error dynamixel_workbench_single_manager is neither a target not built/installed properly

Achllle, 2017-08-02:
I followed the [instructions](http://wiki.ros.org/dynamixel_workbench) to install dynamixel-workbench and dynamixel-workbench-msgs, using both `apt-get` and install from source (both master and kinetic-devel).

`catkin_make` works fine until I create the 'my_dynamixel_workbench_tutorial' package and invoke catkin_make again:
```
CMake Error at /home/achille/Documents/tutorial_ws/devel/share/dynamixel_workbench_single_manager/cmake/dynamixel_workbench_single_managerConfig.cmake:141 (message):
  Project 'my_dynamixel_workbench_tutorial' tried to find library
  'dynamixel_workbench_single_manager'.  The library is neither a target nor
  built/installed properly.  Did you compile project
  'dynamixel_workbench_single_manager'? Did you find_package() it before the
  subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  my_dynamixel_workbench_tutorial/CMakeLists.txt:10 (find_package)
```
I tried the instructions in [issue 54](https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/54) but nothing worked. I tried swapping dynamixel_workbench_single_manager in find_package() in CMakeLists.txt in the tutorial package with dynamixel_controllers from the dynamixel-motor package and that built just fine.

Any solutions to this?

routiful, 2017-08-03:
Hi @Achllle :)

New version of Dynamixel-Workbench isn't downloaded by 'apt-get' command. 
We will upload it as soon as possible. 

So, you have to download it in github.  Using below commands

```
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
```

Before you catkin_make, please purge dynamixel-workbench when you downloaded by 'apt-get' command.

`sudo apt-get purge ros-kinetic-dynamixel-workbench*`

Thanks.
Darby

Achllle, 2017-08-03:
I looked at the issue I mentioned above, ran the purge command andinstalled from source as I mentioned and it still doesn't work.

routiful, 2017-08-03:
Would you show me that error messages in terminal??

Achllle, 2017-08-03:
```
achille@W530:~/Documents/tutorial_ws$ catkin_make
Base path: /home/achille/Documents/tutorial_ws
Source space: /home/achille/Documents/tutorial_ws/src
Build space: /home/achille/Documents/tutorial_ws/build
Devel space: /home/achille/Documents/tutorial_ws/devel
Install space: /home/achille/Documents/tutorial_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/achille/Documents/tutorial_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/achille/Documents/tutorial_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/achille/Documents/rigrade_ros_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/achille/Documents/rigrade_ros_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/achille/Documents/tutorial_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - dynamixel_workbench (metapackage)
-- ~~  - dynamixel_workbench_msgs
-- ~~  - dynamixel_workbench_operators
-- ~~  - dynamixel_workbench_toolbox
-- ~~  - dynamixel_workbench_controllers
-- ~~  - dynamixel_workbench_single_manager
-- ~~  - dynamixel_workbench_single_manager_gui
-- ~~  - my_dynamixel_workbench_tutorial
-- ~~  - actionlib_tutorials
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'dynamixel_workbench'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench)
-- +++ processing catkin package: 'dynamixel_workbench_msgs'
-- ==> add_subdirectory(dynamixel-workbench-msgs/dynamixel_workbench_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- dynamixel_workbench_msgs: 12 messages, 4 services
-- +++ processing catkin package: 'dynamixel_workbench_operators'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_operators)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'dynamixel_workbench_toolbox'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_toolbox)
-- +++ processing catkin package: 'dynamixel_workbench_controllers'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_controllers)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'dynamixel_workbench_single_manager'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_single_manager)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'dynamixel_workbench_single_manager_gui'
-- ==> add_subdirectory(dynamixel-workbench/dynamixel_workbench_single_manager_gui)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'my_dynamixel_workbench_tutorial'
-- ==> add_subdirectory(my_dynamixel_workbench_tutorial)
CMake Error at /home/achille/Documents/tutorial_ws/devel/share/dynamixel_workbench_single_manager/cmake/dynamixel_workbench_single_managerConfig.cmake:141 (message):
  Project 'my_dynamixel_workbench_tutorial' tried to find library
  'dynamixel_workbench_single_manager'.  The library is neither a target nor
  built/installed properly.  Did you compile project
  'dynamixel_workbench_single_manager'? Did you find_package() it before the
  subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  my_dynamixel_workbench_tutorial/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/achille/Documents/tutorial_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/achille/Documents/tutorial_ws/build/CMakeFiles/CMakeError.log".
Makefile:1872: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

```

routiful, 2017-08-04:
I changed tutorial page in [WIKI](http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/CommandLine#preview).

Thanks
Darby

ricardodeazambuja, 2017-08-04:
Hi,

Could you please tell me which branch / tag I should use to make it work (Ubuntu 16.04.2 and ROS Kinetic)? The master branch doesn't compile in my system.
The branch kinetic-devel compiles and the release 0.1.5 compiles, but then the tutorials don't work because they miss some pieces (single_dynamixel_controller, single_dynamixel_controller...).

Thanks!

routiful, 2017-08-04:
Hi, @ricardodeazambuja  :)

**master branch** works fine in Linux Mint Cinnamon and ROS Kinetic.
Would you show me which errors are occurred? 

Thanks
Darby

ricardodeazambuja, 2017-08-04:
Actually, now, the master branch compiled after I purged the ubuntu packages and cloned the dynamixel_workbench_msgs instead. And I just refreshed my browser and used the new tutorial page and it compiled too :)

Thanks!

ricardodeazambuja, 2017-08-04:
Followed the tutorial (http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/CommandLine) and after calling:
$ roslaunch my_dynamixel_workbench_tutorial single_manager.launch
received an ugly "buffer overflow detected" message. Any suggestion?

routiful, 2017-08-04:
You should give proper input voltage to Dynamixel. It can be checked in [E-Manual](http://support.robotis.com/en/)

How about using [GUI tutorial](http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/GUI)?
Is it occurred same message?

Achllle, 2017-08-04:
@ricardodeazambuja Probably a good idea to open a separate issue.

@routiful removing the dependencies worked, but you also removed the dependencies to roscpp and std_msgs. Shouldn't these be added? 
I also had to change the protocol version from 2.0 to 1.0 for it to work on my MX-64Ts

closed 2017-08-04 by Achllle

routiful, 2017-08-07:
To @Achllle 

Due to the tutorial just needs launch file, I removed all dependencies. 
But someone needs to that, I will add 'roscpp' and 'std_msgs'. Thanks to your comments.

And Dynamixel-Workbench don't support protocol version 2.0 for MX Series yet. 
However, we have a plan to support it. Sorry for your inconvenient.

Thanks.
Darby