ROBOTIS-GIT/dynamixel-workbench #60: Controlling multiple dynamixels

jackokaiser, 2017-08-04:
Hello

I have three dynamixel servos (2x MX-28AT and 1x MX-64AT) that I would like to control from /dev/ttyUSB0 at the same time (they are daisy chained).

Using the **single_manager.launch** and the **dynamixel_workbench_single_manager_gui**, I can control each one independently, but when I plug the servos together, the **single_manager.launch** fails to start.

I also noticed **position_control.launch** which uses **dynamixel_multi_driver.cpp**, but this also fails on launch with the error
```
Cant' Load Dynamixel, Please check Parameter
```
It seems that **position_control.launch** expects 2 servos, pan and tilt, whereas I have three plugged in.

Is there an out-of-the box solution to spawn one controller for each dynamixel plugged on /dev/ttyUSB0?

Cheers,
Jacques

jackokaiser, 2017-08-04:
Alright I got the **position_control.launch** to work, the problem was that I did not set the **dynamixel id** of the servo beforehand.

Cheers,
Jacques

routiful, 2017-08-06:
Hi :)

If you have any other questions using Dynamixel-Workbench, please feel free to take a issue ticket

Thanks
Darby

closed 2017-08-06 by routiful