ROBOTIS-GIT/dynamixel-workbench #61: fatal error: dynamixel_workbench_toolbox/dynamixel_multi_driver.h: No such file or directory

sisihi, 2017-08-07:
Hi,
I was just following the http://turtlebot3.robotis.com/en/latest/manipulation.html.
But got the below error:
`/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libcrypto.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
[ 96%] Built target open_manipulator_msgs_generate_messages
[ 96%] Built target turtlebot3_msgs_generate_messages
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `PKCS12_PBE_add@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_basic_verify@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_puts@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_get_peer_certificate@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSLv3_client_method@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_get_id@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `RAND_status@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `MD5_Final@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_set_verify@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ASN1_TIME_print@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_ctrl@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `CONF_modules_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `DES_set_key@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_set_session@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_EXTENSION_get_data@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ERR_error_string_n@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_cert_status_str@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_check_issued@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_set_default_passwd_cb_userdata@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_RESPONSE_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_get_pubkey@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `CRYPTO_malloc@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_use_certificate_chain_file@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSLeay@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `UI_method_get_opener@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_load_crl_file@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ASN1_STRING_type@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ASN1_STRING_data@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `i2t_ASN1_OBJECT@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_get_error@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_NAME_get_entry@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_add_client_CA@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_get_privatekey@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OPENSSL_load_builtin_modules@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `CRYPTO_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `RAND_add@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ASN1_STRING_length@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CIPHER_get_name@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ERR_peek_error@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `sk_num@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_set_next_proto_select_cb@OPENSSL_1.0.1'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_get_serialNumber@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_resp_get0@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_use_PrivateKey_file@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `UI_OpenSSL@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_verify_cert_error_string@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_ctrl_cmd@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `PKCS12_parse@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `MD4_Update@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `RAND_egd@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `d2i_OCSP_RESPONSE@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `UI_get0_user_data@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_get_issuer_name@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_s_mem@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `CRYPTO_cleanup_all_ex_data@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `UI_create_method@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_read@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ERR_remove_thread_state@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_check_private_key@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_printf@OPENSSL_1.0.0'
[... 105 lines cut ...]
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_load_builtin_engines@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_response_status@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_shutdown@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `MD5_Update@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `i2a_ASN1_OBJECT@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_get_verify_result@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_set_msg_callback@OPENSSL_1.0.0'
collect2: error: ld returned 1 exit status
open_manipulator/open_manipulator_position_ctrl/CMakeFiles/position_controller.dir/build.make:206: recipe for target '/home/szpros/catkin_ws/devel/lib/open_manipulator_position_ctrl/position_controller' failed
make[2]: *** [/home/szpros/catkin_ws/devel/lib/open_manipulator_position_ctrl/position_controller] Error 1
CMakeFiles/Makefile2:6322: recipe for target 'open_manipulator/open_manipulator_position_ctrl/CMakeFiles/position_controller.dir/all' failed
make[1]: *** [open_manipulator/open_manipulator_position_ctrl/CMakeFiles/position_controller.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/szpros/catkin_ws/src/open_manipulator/open_manipulator_dynamixel_ctrl/src/dynamixel_controller.cpp:19:0:
/home/szpros/catkin_ws/src/open_manipulator/open_manipulator_dynamixel_ctrl/include/open_manipulator_dynamixel_ctrl/dynamixel_controller.h:28:64: fatal error: dynamixel_workbench_toolbox/dynamixel_multi_driver.h: No such file or directory
compilation terminated.
[... cut ...]
make[2]: *** [open_manipulator/open_manipulator_dynamixel_ctrl/CMakeFiles/dynamixel_controller.dir/src/dynamixel_controller.cpp.o] Error 1
CMakeFiles/Makefile2:3276: recipe for target 'open_manipulator/open_manipulator_dynamixel_ctrl/CMakeFiles/dynamixel_controller.dir/all' failed
make[1]: *** [open_manipulator/open_manipulator_dynamixel_ctrl/CMakeFiles/dynamixel_controller.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
Thanks for the help.

routiful, 2017-08-07:
Hi :)

The debian package are not support new version of Dynamixel-workbench yet.
So, you need to clone master branch in Dynamixel-workbench.
Sorry for inconvenient. We will upload it as soon as possible.

Thanks.
Darby

sisihi, 2017-08-07:
Hi,
I've found http://wiki.ros.org/dynamixel_workbench, git clone and catkinmake,
but got:

[ 84%] Linking CXX executable /home/szpros/catkin_ws/devel/lib/open_manipulator_position_ctrl/position_controller
/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
/usr/bin/cmake: /home/szpros/naoqi/pynaoqi-python2.7-2.1.4.13-linux64/libcrypto.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
/home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:47:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, uint16_t model_number
 ^
In file included from /home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float prot
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float proto
   ^
/home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:60:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, std::string model_nam
 ^
In file included from /home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float prot
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float proto
   ^
/home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp: In member function ‘bool dynamixel_tool::DynamixelTool::getModelName(uint16_t)’:
/home/szpros/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:77:3: error: ‘name_path_’ was not declared in this scope
   name_path_  = ros::package::getPath("dynamixel_workbench_too
   ^
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `PKCS12_PBE_add@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_basic_verify@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_puts@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_get_peer_certificate@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `BIO_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSLv3_client_method@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_get_id@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `RAND_status@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `MD5_Final@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_set_verify@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ASN1_TIME_print@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ENGINE_ctrl@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `CONF_modules_free@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `DES_set_key@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_set_session@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `X509_EXTENSION_get_data@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `ERR_error_string_n@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `OCSP_cert_status_str@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcurl.so.4: undefined reference to `SSL_CTX_free@OPENSSL_1.0.0'
[... cut ...]
make[2]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:4343: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Did I miss something?Thanks for you help.

routiful, 2017-08-07:
Hi :)

You have to purge debian package

$ sudo apt-get purge ros-kinetic-dynamixel-workbench*

Thanks
Darby

closed 2017-09-01 by routiful