ROBOTIS-GIT/dynamixel-workbench #67: Error: There is no stasus packet!

Ash1020, 2017-08-30:
Hi,
I am trying to controll Dynamixel MX-64 (protocol 1.0) by using Ubuntu 14.04 and ROS Indigo, so I followed to dynamixel_workbench_single_manager tutorials.
I succeed to connect to dynamixel and can manage that, but Error had occered suddenly, then I became to be NOT able to control anymore.

The message below,
This function is deprecated. Use 'printf()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
[ERROR] [1504063938.132737040]: [ID] 30, Fail to read!(status_return_level)

This message had displayed and loop with changing the commands what inside (), until I kill and exit this launch file.

What are these errors and how can I possibly fix them?

Thanks

routiful, 2017-09-01:
Hi :)

Did you update MX-64 firmware?

Ash1020, 2017-09-04:
No, I didn't. 
I'll try it.
How can I update that firmware?

routiful, 2017-09-04:
You need to download ROBOPLUS[(link)](http://en.robotis.com/BlueAD/board.php?bbs_id=downloads&mode=view&bbs_no=1152561&page=1&key=&keyword=&sort=&scate=)
It helps update MX-64's firmware and test it in.

Thanks
Darby

orasman, 2017-09-08:
Hi~
you need to check you have connecting more than one dynamixel at the same time. i have same error when i connect one more motor.

routiful, 2017-09-08:
Thats' right. Thank you @orasman :)

closed 2017-09-23 by routiful