ROBOTIS-GIT/dynamixel-workbench #69: Could not find the required component 'dynamixel_workbench_msgs'

Angit16, 2017-09-10:
Hi,
I was trying to install Dynamixel workbench from source in ubuntu 14.04. I use ROS INDIGO. 

```
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "dynamixel_workbench_msgs" with any of the following names:

    dynamixel_workbench_msgsConfig.cmake
    dynamixel_workbench_msgs-config.cmake

  Add the installation prefix of "dynamixel_workbench_msgs" to
  CMAKE_PREFIX_PATH or set "dynamixel_workbench_msgs_DIR" to a directory
  containing one of the above files.  If "dynamixel_workbench_msgs" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_operators/CMakeLists.txt:12 (find_package)

-- Could not find the required component 'dynamixel_workbench_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "dynamixel_workbench_msgs" with any of the following names:

    dynamixel_workbench_msgsConfig.cmake
    dynamixel_workbench_msgs-config.cmake

  Add the installation prefix of "dynamixel_workbench_msgs" to
  CMAKE_PREFIX_PATH or set "dynamixel_workbench_msgs_DIR" to a directory
  containing one of the above files.  If "dynamixel_workbench_msgs" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  dynamixel-workbench/dynamixel_workbench_operators/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/anindya/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/anindya/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
anindya@anindya-Latitude-E7270:~/catkin_ws$ 

```

Can you please have a look?

Angit16, 2017-09-10:
ok, I installed the ROS dependencies and it works fine. Sorry for the question.

routiful, 2017-09-10:
That's ok. If you have any question about dynamixel-workbench, please feel free to ask to me.
Thanks :)

closed 2017-09-10 by routiful