ROBOTIS-GIT/dynamixel-workbench #70: dynamixel workbench is not building in local workspace

gokul-gokz, 2017-09-13:
I have cloned the dynamixel workbench  and dynamixel workbench msgsin local workspace. I have also installed the dependencies for it. But when I run catkin_make command. It throws the following error

> error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, uint16_t model_number)
 ^
In file included from /home/asimov/test_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/asimov/test_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:60:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, std::string model_name)
 ^
In file included from /home/asimov/test_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/asimov/test_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp: In member function ‘bool dynamixel_tool::DynamixelTool::getModelName(uint16_t)’:
/home/asimov/test_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:77:3: error: ‘name_path_’ was not declared in this scope
   name_path_  = ros::package::getPath("dynamixel_workbench_toolbox") + "/dynamixel/model_info.list";
   ^
dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 78%] Linking CXX executable /home/asimov/test_ws/devel/lib/dynamixel_workbench_operators/wheel_operator
[ 79%] Linking CXX executable /home/asimov/test_ws/devel/lib/dynamixel_workbench_operators/joint_operator
[ 79%] Built target wheel_operator
[ 79%] Built target joint_operator
CMakeFiles/Makefile2:2011: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

routiful, 2017-09-13:
Hi :)

If you have installed dynamixel-workbench using 'apt-get' command, you need to get rid of it.

$ sudo apt-get purge ros-kinetic-dynamixel-workbench*

Angit16, 2017-09-13:
Hi, I am using a single dynamixel motor for position and velocity control in ROS. I am not sure if this package will work for that. Should I comment out all the lines concerning to 2nd motor. Imagine if I am using pan only then should I comment out all the tilts?

routiful, 2017-09-14:
Yes. If you want use single dynamixel with position controller, you need to modified code.

gokul-gokz, 2017-09-14:
Yes.Now it is working.Thanks for your quick reply.

closed 2017-09-14 by gokul-gokz