ROBOTIS-GIT/dynamixel-workbench #71: Failed to read!

gokul-gokz, 2017-09-14:
I have connected 2 dynamixel motors(H54-100-S500-R and H54-200-S500-R) of protocol 2.0. I have set the baud rate to 1Mbps.  Then I set the parameters in position_control.launch accordingly. But when I launch the node, it throws error:

> [ERROR] [1505359515.944733454]: Cant' Load Dynamixel, Please check Parameter
[TxRxResult] There is no status packet!
[ERROR] [1505359515.964138852]: [ID] 1, Fail to read!(torque_enable)
[TxRxResult] There is no status packet!
[ERROR] [1505359515.982326088]: [ID] 1, Fail to read!(present_position)
[TxRxResult] There is no status packet!
[ERROR] [1505359516.000509470]: [ID] 1, Fail to read!(goal_position)
[TxRxResult] There is no status packet!
[ERROR] [1505359516.020910637]: [ID] 1, Fail to read!(goal_velocity)

I have also tried launching the node 'single manager', but it also showed the same kind of error. 
And the firmware of the motors are in latest version which I have checked it.
I have cloned the dynamixel-workbench locally.
How to fix this issue?

routiful, 2017-09-14:
Hi :)

Did you supply the proper voltage(24V) to the Dynamixel PRO?

gokul-gokz, 2017-09-14:
Ya.the power supplied 24v and in windows client(R+manager), I am able to control the motors.

gokul-gokz, 2017-09-19:
This issue have been fixed. It was an wiring issue in the manipulator.

routiful, 2017-09-23:
I don't know how to set your development environments.
But, if you happen that error again please notice me any time

Thanks.

closed 2017-09-23 by routiful