ROBOTIS-GIT/dynamixel-workbench #72: Latency of dynamixel RX-24F

Angit16, 2017-09-14:
Hi,
I have looked into many packages other than this one regarding how to play with dynamixel servos.
I am still unclear about the latency of individual dynamixel. Can you please provide some information?
In addition, I want to know about return delay as well. According to the dynamixel manuals, based on the Buadrate return delay time changes. How is it related to different size of data packets that one would like to read or write?

thanks,

routiful, 2017-09-23:
Sorry. I am business trip now. I will check your issue as soon as possible.

routiful, 2017-10-10:
Hi :) 

You can get an information about return delay time in [E-Manual](http://support.robotis.com/en/)
Following below 
<kbd>http://support.robotis.com/en/</kbd> -> <kbd>Product Information</kbd> -> <kbd>Actuator</kbd> -> <kbd>Dynamixel X</kbd> -> <kbd>XM Series</kbd> -> <kbd>XM430-W350</kbd> -> <kbd>Return Delay Time</kbd>

Thanks
Darby

closed 2017-11-05 by routiful