ROBOTIS-GIT/dynamixel-workbench #73: leakage found while running single_manager.launch

Angit16, 2017-09-22:
Hi,

I also observed that there is a leakage in the codes while running the single_manager.launch. 
After the execution of `...wait for seconds` in dynamixel_driver.cpp, `...Succeeded to find dynamixel` in the single_dynamixel_monitor.cpp, it enters into single_dynamixel.controller.cpp. I found this by using print statement in the `while` loop in the `ros::Rate` of  both the codes.  Why is it so? Also, even if I don't enter any command through keyboard,  the process switches between the `while` loops of two `ros::Rate` . Why is it so? I am confused?

routiful, 2017-09-23:
Sorry. I am business trip now. I will check your issue as soon as possible.

routiful, 2017-10-10:
Hi :)

single_dynamixel_controller(controller) and single_dynamixel_monitor(monitor) has relationship between server and client.
controlLoop function in controller waits enter key(kbhit() function) to send commands.
Therefore, you can see print statement during scan dynamixel even any commands.

Thanks
Darby

closed 2017-11-05 by routiful