ROBOTIS-GIT/dynamixel-workbench #75: WheelCommand.h: No such file or directory

peterheim1, 2017-10-08:
when I run catkin_make I get the following error
system is Rasbery pi 3 ubuntu mate 16.4 ros kinetic from debs all up to date 
` Building CXX object dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_driver.cpp.o
/home/pi/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_operators/src/wheel_operator.cpp:23:51: fatal error: dynamixel_workbench_msgs/WheelCommand.h: No such file or directory
/home/pi/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_operators/src/joint_operator.cpp:20:51: fatal error: dynamixel_workbench_msgs/JointCommand.h: No such file or directory
compilation terminated.
compilation terminated.
dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/src/joint_operator.cpp.o' failed
dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/src/wheel_operator.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/src/joint_operator.cpp.o] Error 1
make[2]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/src/wheel_operator.cpp.o] Error 1
CMakeFiles/Makefile2:2731: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/joint_operator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:3139: recipe for target 'dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_operators/CMakeFiles/wheel_operator.dir/all] Error 2
`

routiful, 2017-10-11:
Hi :)

Latest Dynamixel-workbench is not updated debian package yet. 
So, please download source code in github.

Purge debian packages
`sudo apt-get purge ros-kinetic-dynamixel-workbench*`

Dependent packages
```
sudo apt-get install ros-kinetic-dynamixel-sdk
sudo apt-get install ros-kinetic-qt-build
```
Clone source code
```
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
```
Thanks
Darby

tfoote, 2017-10-16:
This is failing to build on the buildfarm too: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__dynamixel_workbench_operators__ubuntu_xenial_amd64__binary/71/console

routiful, 2017-10-18:
Hi @tfoote :)

dynamixel-workbench-msgs package need to be update.
We will be request it soon.

Thanks
Darby

routiful, 2018-02-28:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-28 by routiful