ROBOTIS-GIT/dynamixel-workbench #76: Position controller for 1 single Dynamixel

christophebedard, 2017-10-13:
I want to control 3 Dynamixels, but the `position_control` node is set up for pan+tilt. I want one controller that can be instantiated for every Dynamixel I have (given different/unique IDs). `single_dynamixel_controller` doesn't seem like it's made for this either.

I can't seem to find what I want. Does it exist?

christophebedard, 2017-10-13:
#60 seemed to ask for the same thing
> Is there an out-of-the box solution to spawn one controller for each dynamixel plugged on /dev/ttyUSB0?

routiful, 2017-10-25:
[OpenManipulator](https://github.com/ROBOTIS-GIT/open_manipulator/tree/master/open_manipulator_dynamixel_ctrl) uses dynamixel workbench to control 5 dynamixel.
This example will help you.

Thanks
Darby

christophebedard, 2017-10-26:
Looks great, thank you!

closed 2017-10-26 by christophebedard