ROBOTIS-GIT/dynamixel-workbench #77: dynamixel_workbench_toolbox fails to build in Kinetic

tfoote, 2017-10-27:
It does not build on the buildfarm and looks to be an error in the cmake configure step. Quickly looking at the traceback my guess is that a python dependency is not properly declared. 

http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__dynamixel_workbench_toolbox__ubuntu_xenial_amd64__binary/106/console
```
23:15:26 GTEST_INCLUDE_DIR-ADVANCED:INTERNAL=1
23:15:26 //ADVANCED property for variable: GTEST_LIBRARY
23:15:26 GTEST_LIBRARY-ADVANCED:INTERNAL=1
23:15:26 //ADVANCED property for variable: GTEST_LIBRARY_DEBUG
23:15:26 GTEST_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
23:15:26 //ADVANCED property for variable: GTEST_MAIN_LIBRARY
23:15:26 GTEST_MAIN_LIBRARY-ADVANCED:INTERNAL=1
23:15:26 //ADVANCED property for variable: GTEST_MAIN_LIBRARY_DEBUG
23:15:26 GTEST_MAIN_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
23:15:26 //ADVANCED property for variable: PYTHON_EXECUTABLE
23:15:26 PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
23:15:26 //This needs to be in PYTHONPATH when 'setup.py install' is called.
23:15:26 //  And it needs to match.  But setuptools won't tell us where
23:15:26 // it will install things.
23:15:26 PYTHON_INSTALL_DIR:INTERNAL=lib/python2.7/dist-packages
23:15:26 
23:15:26 dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_PREFIX_PATH=/opt/ros/kinetic returned exit code 1
23:15:26 	cd /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.1.6
23:15:26 debian/rules:26: recipe for target 'override_dh_auto_configure' failed
23:15:26 make[1]: *** [override_dh_auto_configure] Error 255
23:15:26 make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.1.6'
23:15:26 debian/rules:23: recipe for target 'build' failed
23:15:26 make: *** [build] Error 2
23:15:26 dpkg-buildpackage: error: debian/rules build gave error exit status 2
23:15:26 E: Building failed
23:15:26 Traceback (most recent call last):
23:15:26   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
23:15:26     subprocess.check_call(cmd, cwd=source_dir)
23:15:26   File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
23:15:26     raise CalledProcessError(retcode, cmd)
23:15:26 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-kinetic-dynamixel-workbench-toolbox']' returned non-zero exit status 1
23:15:26 # END SUBSECTION
```

routiful, 2017-10-30:
Hi :)

I think this error because of roslib.
I add python dependency. I wish it will be build completely.

Thanks.
Darby

tfoote, 2017-11-01:
It made it further however it looks like there's a invald install rule now: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__dynamixel_workbench_operators__ubuntu_xenial_amd64__binary/89/console

```
11:35:17 CMake Error at cmake_install.cmake:91 (file):
11:35:17   file INSTALL cannot find
11:35:17   "/tmp/binarydeb/ros-kinetic-dynamixel-workbench-operators-0.1.7/include/dynamixel_workbench_operators".
11:35:17 
11:35:17 
```

routiful, 2017-11-01:
Sorry, my fault.
I deleted INSTALL line.

Thanks
Darby

tfoote, 2017-11-01:
Operators is building correctly now, but the toolbox appears to have a similar issue: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__dynamixel_workbench_toolbox__ubuntu_xenial_amd64__binary/114/console

```
GTEST_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: GTEST_LIBRARY_DEBUG
GTEST_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: GTEST_MAIN_LIBRARY
GTEST_MAIN_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: GTEST_MAIN_LIBRARY_DEBUG
GTEST_MAIN_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_EXECUTABLE
PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
//This needs to be in PYTHONPATH when 'setup.py install' is called.
//  And it needs to match.  But setuptools won't tell us where
// it will install things.
PYTHON_INSTALL_DIR:INTERNAL=lib/python2.7/dist-packages

dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_PREFIX_PATH=/opt/ros/kinetic returned exit code 1
	cd /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.1.8
debian/rules:26: recipe for target 'override_dh_auto_configure' failed
make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.1.8'
make[1]: *** [override_dh_auto_configure] Error 255
debian/rules:23: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
E: Building failed
Traceback (most recent call last):
  File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
    subprocess.check_call(cmd, cwd=source_dir)
  File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-kinetic-dynamixel-workbench-toolbox']' returned non-zero exit status 1
# END SUBSECTION
```

routiful, 2017-11-03:
Sorry again.
I modified CMakeLists in toolbox because of below error. Moreover, I checked other packages in dynamixel-workbench.

Thanks
Darby

```
00:59:08 CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
00:59:08   Could not find a package configuration file provided by "dynamixel_sdk"
00:59:08   with any of the following names:
00:59:08 
00:59:08     dynamixel_sdkConfig.cmake
00:59:08     dynamixel_sdk-config.cmake
00:59:08 
00:59:08   Add the installation prefix of "dynamixel_sdk" to CMAKE_PREFIX_PATH or set
00:59:08   "dynamixel_sdk_DIR" to a directory containing one of the above files.  If
00:59:08   "dynamixel_sdk" provides a separate development package or SDK, be sure it
00:59:08   has been installed.
00:59:08 Call Stack (most recent call first):
00:59:08   CMakeLists.txt:12 (find_package)
```

robotpilot, 2018-02-28:
Solved! ref issues: https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/117

closed 2018-02-28 by robotpilot