ROBOTIS-GIT/dynamixel-workbench #90: Velocity control tutorial fails to move

philko4711, 2017-11-08:
Dear Ladies and Gentlemen,
I tried out the dynamixel_workbench tutorial "Dynamixel workbench velocity controller with wheel command". I created everything according to the tutorial. After I start the launch file both servos are found and enabled (you can't move the motors by hand anymore). 
When I send a velocity command the motors are not moving...do you have any idea, what I might done wrong? The "Dynamixel workbench position controller with joint command" tutorial works fine.

Thank you very much. 

Kind regards, 

Philipp

ROBOTIS-Leon, 2017-12-04:
@philko4711 

Hi,

did you change control mode of the motors from position control mode into velocity control mode?

philko4711, 2017-12-04:
Hi,
the servos are both set to joint mode in roboPlus software. Do I need to set them to wheel mode?

ROBOTIS-Leon, 2017-12-04:
@philko4711 

Hi, again.

Yes, you should change it into wheel mode

philko4711, 2017-12-04:
When I set them to wheel mode, it deletes the angle thresholds. When I try and reset them, the mode gets automatically changed into joint mode.
Is it possible to set minimal or maximal angles in wheel mode?

philko4711, 2017-12-04:
Hi,
yes I thought so, too. 
Exactly. We have a robot with a sensor head on top. The yaw and pitch angle are controlled by an analogue stick of a gamepad. The operator shall set the speed with the stick. If he lets it go, the head shall keep the last pose. 
We are very happy with the new ROS packages but we would like to use the velocity control mode. We achieved our goal in position control as well but it was a bit difficult. Is there any way to use the velocity control without being in wheel mode?

ROBOTIS-Leon, 2017-12-04:
@philko4711 

What kinda motor are you using now?

philko4711, 2017-12-04:
We are using two dynamixel mx-28

ROBOTIS-Leon, 2017-12-04:
@philko4711 

Upgrade its firmware to No.41 or above (If you can change its protocol to 2.0, it means your dynamixel has already No.41 or above, you can check it by roboplus manager 2.0)
 
then, it might be having a devided system: its mode selection might be separated with angular limitation. 

nevertheless, the motor will disregard the angular limitation when it is wheel mode. the thing is; you can catch its position out even when it is running as wheel mode. You can control it by wheel mode, but you can read its position, so you can handle its angle at the same time

philko4711, 2017-12-04:
We just discussed this case. You mean, we should limit the angles by our software?! 
In this case, we prefer to use our already working version. It is a bit complicated but it works fine. We have very expensive sensors and are afraid, that an error in our software might cause a defect.

We are very happy with the new ros packages by Robotis. They solved a lot of problems. Thanks for that

ROBOTIS-Leon, 2017-12-04:
@philko4711 

I hope your work goes well.

Thank you and please ask if you have further questions

closed 2017-12-04 by ROBOTIS-Leon