ROBOTIS-GIT/dynamixel-workbench #92: Not working

swilker01, 2017-11-20:
Has anyone seen this, probably something I'm forgetting, stuck...

roslaunch my_dynamixel_workbench_tutorial single_monitor.launch
... logging to /home/swilker/.ros/log/d1e01a02-cda2-11e7-b710-dc5360ceec67/roslaunch-Europa-9033.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Europa:44255/

SUMMARY
========

PARAMETERS
 * /baud_rate: 100000
 * /device_name: /dev/ttyUSB0
 * /ping: False
 * /ping_id: 1
 * /protocol_version: 2.0
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
    single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

auto-starting new master
process[master]: started with pid [9045]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d1e01a02-cda2-11e7-b710-dc5360ceec67
process[rosout-1]: started with pid [9058]
started core service [/rosout]
ERROR: cannot launch node of type [dynamixel_workbench_single_manager/single_dynamixel_monitor]: dynamixel_workbench_single_manager
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/swilker/catkin_ws/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks
ERROR: cannot launch node of type [dynamixel_workbench_single_manager/single_dynamixel_controller]: dynamixel_workbench_single_manager
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/swilker/catkin_ws/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks

Torugo, 2017-11-27:
I'm facing the same problem. Did you find a way to solve it?

ROBOTIS-Leon, 2017-12-06:
@swilker01 @Torugo 

Hi, 
I hope that you are still currently working on this.

Have you guys tried even when the IP address is localhost?

Torugo, 2017-12-06:
I can't build the package from git, and aptitude can't find it

```
sudo apt-get install ros-kinetic-dynamixel-workbench
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-dynamixel-workbench
```

ROBOTIS-Leon, 2017-12-07:
@Torugo 

Hi, 

at this moment, it's better cloning the package by:

``` bash
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
```

the package in the server will be soon recovered.

Thank you.

Torugo, 2017-12-08:
It doesn't build, as I said:
`/home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:47:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, uint16_t model_number)
 ^
In file included from /home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:60:1: error: prototype for ‘dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string)’ does not match any in class ‘dynamixel_tool::DynamixelTool’
 DynamixelTool::DynamixelTool(uint8_t id, std::string model_name)
 ^
In file included from /home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:19:0:
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:64:7: error: candidates are: dynamixel_tool::DynamixelTool::DynamixelTool(const dynamixel_tool::DynamixelTool&)
 class DynamixelTool
       ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:92:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, std::__cxx11::string, float)
   DynamixelTool(uint8_t id, std::string model_name, float protocol_version);
   ^
/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_tool.h:91:3: error:                 dynamixel_tool::DynamixelTool::DynamixelTool(uint8_t, uint16_t, float)
   DynamixelTool(uint8_t id, uint16_t model_number, float protocol_version);
   ^
/home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp: In member function ‘bool dynamixel_tool::DynamixelTool::getModelName(uint16_t)’:
/home/vitor/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_tool.cpp:77:3: error: ‘name_path_’ was not declared in this scope
   name_path_  = ros::package::getPath("dynamixel_workbench_toolbox") + "/dynamixel/model_info.list";
   ^
dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/build.make:62: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o' failed
make[2]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/src/dynamixel_tool.cpp.o] Error 1
CMakeFiles/Makefile2:3348: recipe for target 'dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all' failed
make[1]: *** [dynamixel-workbench/dynamixel_workbench_toolbox/CMakeFiles/dynamixel_workbench_toolbox.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
`

routiful, 2017-12-11:
Hi @Torugo :)

Sorry for your inconvenience.
The package will be update next month. 
If you download(clone) package in github, you would have to purge installed package.

`sudo apt-get purge ros-{ros-distro}-dynamixel-workbench*`

Above instruction will be help you.

Thanks
Darby

closed 2018-02-13 by routiful

Ash-S, 2019-11-28:
Hi @LeonJung / @routiful 
I am developing my application on RPi connected with number of dynamixels bought from ROBOTIS KR based on ROS Kinetic platform. I have installed package clone as per shared link.
_git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git_
but failed to build it in ROS Kinetic. I failed to found proper solution of it. I request you to please help me to resolve this issue.  waiting for your kind reply~ Thanks!!