ROBOTIS-GIT/dynamixel-workbench #93: support for XL-320

sonelu, 2017-11-20:
Looks like the XL-320 is not included in the available servos. It should be easy to create an XL_320.device in dynamixel_workbench_toolbox/dynamixel/models/XL but I feel the dynamixel-workbench-msgs/dynamixel_workbench_msgs/msg/XL will not be suitable for XL-320 too. They might have to be split with their own topic.

I've done a quick check with an XL_320.device file, and I can connect to the servo using the single_manager, I can move the servo (by issuing goal_position commands) but I also get a lot of read errors which I don't think are from the interface.

ROBOTIS-Leon, 2017-12-06:
@sonelu 

Hi,

that's a good point. XL-320 is actually a XL series model, but it doesn't stand with the other XL series by now. It should be, unfortunately, placed to the AX, MX which has almost the same control table.

Any ideas, please feedback us

routiful, 2017-12-11:
Hi @sonelu :)

XL-320 is not included in latest Dynamixel Workbench. It will be include next version which will be released next month.
If you show me which error messages occurred, it might be helpful to me. 
Moreover, Input voltage of XL-320 is recommended 7.4V. You need to check it before move the [motor](http://emanual.robotis.com/docs/en/dxl/x/xl320/).

Thanks
Darby

routiful, 2018-02-13:
Hello

Latest Dynamixel Workbench is released. It supports include XL-320.
If you need to any supports, please feel free to open this issue.

Thanks
Darby

closed 2018-02-13 by routiful

LasseBoerresen, 2018-02-19:
Sorry, maybe I am blind, but I cannot see anywhere that the next Dynamixel Workbench has been released. Here on github it is still 0.1.9 and when purging and then running 'sudo apt-get install ros-kinetic-dynamixel-workbench'  on ubuntu 16.4 Xenial.

routiful, 2018-02-20:
Hello @LasseBoerresen :)

I'm sorry. You are not blind. 
We have released [Dynamixel-Workbench 0.2.0](https://github.com/ROBOTIS-GIT/dynamixel-workbench/releases) in this morning.

Install it and feel free to ask anything about it.

Best regards
Darby

Tibeo, 2019-08-07:
Hello @routiful, 

Thanks for your time. I can use Dynamixel Wizard 2.0 to connect XL320 with U2D2 and drive it. The port name is "/dev/ttyUSB0" and I set the baud rate as 57600 in Dynamixel Wizard. However, Dynamixel workbench could not find any dynamixels with the following command
"rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0".

The result comes as follows.
[ INFO] [1565165026.078157064]: Succeed to init(9600)
[ INFO] [1565165026.078203685]: Wait for scanning...
[ INFO] [1565165047.861067288]: Find 0 Dynamixels
[ INFO] [1565165047.861530181]: Succeed to init(57600)
[ INFO] [1565165047.861587558]: Wait for scanning...
[ INFO] [1565165065.866423184]: Find 0 Dynamixels
[ INFO] [1565165065.866775165]: Succeed to init(115200)
[ INFO] [1565165065.866819142]: Wait for scanning...
[ INFO] [1565165083.526925239]: Find 0 Dynamixels
[ INFO] [1565165083.527278402]: Succeed to init(1000000)
[ INFO] [1565165083.527313947]: Wait for scanning...
[ INFO] [1565165100.855615502]: Find 0 Dynamixels
[ INFO] [1565165100.855975034]: Succeed to init(2000000)
[ INFO] [1565165100.856015821]: Wait for scanning...
[ INFO] [1565165118.160769416]: Find 0 Dynamixels
[ INFO] [1565165118.161240482]: Succeed to init(3000000)
[ INFO] [1565165118.161289975]: Wait for scanning...
[ INFO] [1565165135.460835977]: Find 0 Dynamixels
[ INFO] [1565165135.461387152]: Succeed to init(4000000)
[ INFO] [1565165135.461438061]: Wait for scanning...
[ INFO] [1565165152.756168884]: Find 0 Dynamixels.

I used this command to find XH430-W210. It did work. Could I know whether current Dynamixel workbench supports XL320? Thank you.