ROBOTIS-GIT/dynamixel-workbench #94: Please Check USB Port authorization and Baudrate [ex : 57600, 115200, 1000000, 3000000]

YuehChuan, 2017-12-15:
Hello Dorby and Leon,
I'm using  AX-12A with U2D2 new connector.
This is ma hardware configuration.
[image]

follow this.
http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/GUI
Build from source and when I want open  GUI to  set motor ID . 

shows
[image]

```bash
process[single_dynamixel_monitor-2]: started with pid [7149]
[ INFO] [1513363628.617394991]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1513363628.624420877]: Succeeded to change the baudrate(57600)!
[ INFO] [1513363628.624483780]: Scan Dynamixel(ID: 1~253) Using Protocol 2.0

[ INFO] [1513363628.624540752]: ...wait for seconds
[ WARN] [1513363633.662755955]: Please Check USB Port authorization and
[ WARN] [1513363633.662802262]: Baudrate [ex : 57600, 115200, 1000000, 3000000]
[ERROR] [1513363633.662828001]: ...Failed to find dynamixel!
================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
process has died [pid 7149, exit code -11, cmd /home/schwarm/techman/catkin_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/schwarm/.ros/log/5a270a90-e1c8-11e7-8e7e-0811961618c4/single_dynamixel_monitor-2.log].
log file: /home/schwarm/.ros/log/5a270a90-e1c8-11e7-8e7e-0811961618c4/single_dynamixel_monitor-2*.log
Initiating shutdown!
================================================================================
[single_dynamixel_monitor-2] killing on exit

```

Any hint?
TQ in advance!

ROBOTIS-Leon, 2017-12-18:
@YuehChuan 

Long time no see,

you are using AX series, right?
it can't be controlled with `Protocol 2.0 comm packet`

Change here to `1.0`:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/399cceb071283e58055a7f0db14a5b6636892a22/dynamixel_workbench_single_manager/launch/single_manager.launch#L7

and check also if the baudrate and id are set properly.

YuehChuan, 2017-12-18:
yes, AX-12A. I will try it. TQ!

YuehChuan, 2017-12-30:
I get no luck
[image]
and this is running dynamixelSDK read_write example
[image]

YuehChuan, 2017-12-30:
get no luck
[image]
and this is SDK 
[image]
read_write example output

routiful, 2018-01-15:
Hi @YuehChuan 

Sorry for late answer. 
Did you launch it different baudrate??
AX series has default 1M baudrate.

ROBOTIS-Leon, 2018-01-15:
@routiful 

He think that his AX has a problem,
let's wait for his response

YuehChuan, 2018-01-15:
@routiful  @ROBOTIS-Leon  TQ! I will update here if any progress!

routiful, 2018-02-13:
Hi @YuehChuan 

Do you solve this problem??

routiful, 2018-02-28:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-28 by routiful

RishavGH, 2018-08-24:
I am having the same problem. Please tell me how to solve this problem. I have tried after changing the protocol but it doesn't work the next time I connect the motor.