ROBOTIS-GIT/dynamixel-workbench #100: position_control groupSyncWrite addparam failed

ahmohamed1, 2018-01-18:
Hi , 

I am using ubuntu 16.04 with ROS Kinect.

I am trying to use the new Dynamixel XL430, 
when I run roslaunch my_dynamixel_workbench_tutorial position_control.launch

I get this error :
`... logging to /home/abdulla/.ros/log/1cf88f90-fc80-11e7-afb1-6045cb73f98d/roslaunch-PC-29810.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PC:46061/

SUMMARY
========

PARAMETERS
 * /baud_rate: 57600
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /scan_range: 10

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

auto-starting new master
process[master]: started with pid [29820]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1cf88f90-fc80-11e7-afb1-6045cb73f98d
process[rosout-1]: started with pid [29834]
started core service [/rosout]
process[position_control-2]: started with pid [29843]
[ INFO] [1516301283.269574175]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1516301283.270010627]: Succeeded to change the baudrate(57600)!
[ INFO] [1516301283.317941676]: -----------------------------------------------------------------------
[ INFO] [1516301283.317959145]:   dynamixel_workbench controller; position control example(Pan & Tilt) 
[ INFO] [1516301283.317967067]: -----------------------------------------------------------------------
[ INFO] [1516301283.317975814]: PAN MOTOR INFO
[ INFO] [1516301283.317983910]: ID    : 1
[ INFO] [1516301283.317991538]: MODEL : XL430_W250
[ INFO] [1516301283.317998468]:  
[ INFO] [1516301283.318005471]: TILT MOTOR INFO
[ INFO] [1516301283.318012563]: ID    : 1
[ INFO] [1516301283.318019038]: MODEL : XL430_W250
[ INFO] [1516301283.318027296]:  
[ INFO] [1516301283.318034213]: Profile Velocity     : 0
[ INFO] [1516301283.318043275]: Profile Acceleration : 0
[ INFO] [1516301283.318050872]: -----------------------------------------------------------------------
[ERROR] [1516301283.318066433]: [ID:001] groupSyncWrite addparam failed
[ERROR] [1516301283.318073184]: SyncWrite Torque Failed!
[ERROR] [1516301283.318080461]: [ID:001] groupSyncWrite addparam failed
[ERROR] [1516301283.318086041]: SyncWrite Profile Velocity Failed!
`
One more things I try to detect AX-12A but the software cannot detect the motor.

routiful, 2018-01-19:
Hello :)

The new version was released last day.
You can download it in master branch.
Before you launch new version, please purge you had.

`$ sudo apt-get purge ros-kinetic-dynamixel-workbench*`

Thanks
Darby

ahmohamed1, 2018-01-19:
Thank you @routiful,
That sort of the problem now I can control the motor. 

There is the other issue that I cannot run ax-12a using this package even when I run it alone.

I have the following error:

`roslaunch my_dynamixel_workbench_tutorial position_control.launch 
... logging to /home/abdulla/.ros/log/686ef198-fcef-11e7-b913-6045cb73f98d/roslaunch-PC-10049.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PC:38257/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /scan_range: 10

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

auto-starting new master
process[master]: started with pid [10059]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 686ef198-fcef-11e7-b913-6045cb73f98d
process[rosout-1]: started with pid [10072]
started core service [/rosout]
process[position_control-2]: started with pid [10080]
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : AX-12A
ID      : 3

MODEL   : XL430-W250
ID      : 1

MODEL   : XL430-W250
ID      : 2

-----------------------------------------------------------------------
[position_control-2] process has died [pid 10080, exit code -11, cmd /home/abdulla/dev/Active-stereo-Vision-Platform/ros/devel/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/abdulla/.ros/log/686ef198-fcef-11e7-b913-6045cb73f98d/position_control-2.log].
log file: /home/abdulla/.ros/log/686ef198-fcef-11e7-b913-6045cb73f98d/position_control-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

routiful, 2018-01-22:
Hi :)

AX-12A and XL430-W250 have different name of control table item for getting information for velocity.
dynamixel-workbench provides examples compatible with dynamixel using protocol 2.0.
So you need to change some code for using AX-12A. I give a sample code for you.

```
if (strncmp(dxl_wb_->getModelName(3), "AX", 2))
{
    dynamixel_state[index].present_velocity    = dxl_wb_->itemRead(dxl_id_[index], "Present_Speed"); 
    dynamixel_state[index].goal_velocity       = dxl_wb_->itemRead(dxl_id_[index], "Moving_Speed");   
}
```

Best regards
Darby

ahmohamed1, 2018-01-22:
Hi Darby,

Thank you for your reply. 

is it possible to use both AX-12 and XL-430-W250 in one port?
when I run the position_controller the program fails at this point 

`dxl_wb_->jointMode(dxl_id_[index], profile_velocity, profile_acceleration);
`
Regards,
Abdulla

routiful, 2018-01-23:
Oh, my mistake. Sorry for confusing.
In fact, AX is controlled by protocol 1.0 and X is operated by protocol 2.0.

Dynamixel-workbench is structured to have only one protocol handler. Hence, you don't use it multiple Dynamixel has different protocol version.

If you want to use it simultaneously, you have to run [multi port example](http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials/MultiPort) or using [Dynamixel SDK example.](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_protocol_combined/#cpp-protocol-combined)

Thanks
Darby

routiful, 2018-02-28:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-28 by routiful

alexisutc, 2019-11-27:
Hello,

i'm facing following problem :
[TxRxResult] Incorrect status packet!

I'm using the dynamixel workbench in ros. So basically i start the node and i got this :

[ INFO] [1574858977.352412572]: Name : deux, ID : 2, Model Number : 12 [ INFO] [1574858977.356064870]: Name : quatre, ID : 4, Model Number : 12 [ INFO] [1574858977.359673210]: Name : trois, ID : 3, Model Number : 12 [ INFO] [1574858977.362450403]: Name : un, ID : 1, Model Number : 12 [ INFO] [1574858977.500291124]: [DynamixelDriver] Succeeded to add sync write handler [ INFO] [1574858977.500466176]: [DynamixelDriver] Succeeded to add sync write handler [ERROR] [1574858978.004020952]: [TxRxResult] Incorrect status packet! [ERROR] [1574858978.072226858]: [TxRxResult] There is no status packet! [ERROR] [1574858978.145851471]: [TxRxResult] Incorrect status packet! [ERROR] [1574858981.353259468]: [TxRxResult] Incorrect status packet! [ERROR] [1574858982.220028446]: [TxRxResult] Incorrect status packet!

config :
Raspberry pi (and Navio2) working with raspbian
also I dont have
cat: /sys/bus/usb/devices/usb1/latency_timer: No such file or directory

I'm using the USB2AX since it is for a drone and also is the smallest one

Also sometimes it crashes with
groupSyncWrite addparam failed