ROBOTIS-GIT/dynamixel-workbench #102: Protocol 1.0 & 2.0 combined in one port

andreaskargakos, 2018-01-23:
Hi, 
I am using 2 XM430-W210 (protocol 2.0) for velocity control and one AX-12A (protocol 1.0) for position control. The problem is that i can not use them with this configuration since its not supported (?) from the dynamixel-workbench package.

I have also seen the example with the sdk : http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_protocol_combined/#cpp-protocol-combined

**_Is it possible to add this kind of driver in the dynamixel-workbench package,  like the multi_driver and the multi_port one is?_** 

Is there any other solution at the moment using the workbench package?

routiful, 2018-01-25:
Hi @andreaskargakos :)

Dynamixel-workbench is not support protocol combined yet. 
If i taking similar issues, i would consider it next update.

Dynamixel-workbench is updated. You can download it in master branch.

Thanks
Darby

andreaskargakos, 2018-01-25:
@routiful i can see that i can change the protocol version of my XM430-W210 to protocol 1.0 using the dynamixel-workbench-single-manager-gui, but then it fails to launch the velocity_control.launch
Am i doing something wrong? Why i m not able to work the XM with the protocol 1.0 set by the gui?

routiful, 2018-01-31:
Hi :) 

Sorry for late comment.
I updated protocol combine function in dynamixel-workbench-toolbox.
You can download it in master branch.
After use it please comment how it works.

Thanks
Darby

routiful, 2018-02-05:
Hi :)

I tested velocity controller with two XM430-W210s(protocol version 1.0). It is working well. 
In fact, velocity controller can ping two dynamixels only.
But if you want to use two XM430-W210s(velocity control)and one AX(position control), you have to make your own package to use dynaixel-workbench toolbox.

Best regards
Darby

routiful, 2018-02-28:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-28 by routiful

KvalheimRacing, 2018-07-12:
I didn't relly get what happened here. @routiful did you fix communications partly since you are talking about that velocity and position control combined is not working?
I have the same problem, have this 6DOF [crustcrawler](http://www.crustcrawler.com) with MX64 (2.0) servos, and on the end, a dual gripper consistent of two AX-18A (1.0) servos. I just want a controller for the gripper (position), don't think these servos has torque control. Can dynamixel-workbench help me with that, or is there not support for linking through so many MX64 2.0's?

routiful, 2018-07-13:
Hi @KvalheimRacing 

Unfortunately position_control.launch file only supports single packetHandler for protocol version 1.0 or 2.0. So you are supposed to set MX-64 to MX-64(2.0) to use AX-18A simultaneously.

KvalheimRacing, 2018-07-13:
I am confused, AX-18A is protocol 1.0. What did you mean by MX64 to MX64(2.0), link two different MX64 servos?

routiful, 2018-07-13:
Sorry, I gave you wrong answer. 

In fact, [AX](http://emanual.robotis.com/docs/en/dxl/ax/ax-18a/) only supports protocol version 1.0 and [MX](http://emanual.robotis.com/docs/en/dxl/mx/mx-64/) supports protocol version 1.0 too. But [MX(2.0)](http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/) can support protocol version 1.0 and 2.0.

Before you launch position control example, you supposed to set same protocol version to AX and MX (NOT MX(2.0))

KvalheimRacing, 2018-07-13:
Yes that is my point exactly. I have this 6DOF [crustcrawler](http://www.crustcrawler.com) with MX64 (2.0) servos, and on the end, a dual gripper consistent of two AX-18A (1.0) servos. I am to make a controller for the gripper (position) that talks 1.0 to the rest of the 2.0 servos.  Can dynamixel-workbench help me with that?

routiful, 2018-07-13:
Unfortunately, Dynamixel workbench has not supported change or upgrade firmware of Dynamixel.
You are supposed to use [Dynamixel Wizard](http://emanual.robotis.com/docs/en/software/rplus1/dynamixel_wizard/#introduction)

KvalheimRacing, 2018-07-13:
I don't want to change the firmware. I want to communicate bewteen a dynamixel servo using 1.0 protocol and another servo using 2.0 protocol. Since http://wiki.ros.org/dynamixel_workbench shows support for both AX-18F/18A, (which is 1.0) and MX-64(2.0) (which is 2.0) i guessed that you had some sort of communication bridge between the two protocols.

routiful, 2018-07-16:
Of course, using Dynamixel-Workbench, you can use both of them simultaneously. 
However, now the example you can shown is to provide how to control two motors have different protocol by using different ports. If you want to use them one ports, you have to make your own code to deal with them. 

If you have any question when you develop the code, please ask me feel free.
I will update the example near future.

KvalheimRacing, 2018-07-17:
I see, thanks for answering. I think the solution is just to change to servos that supports 2.0

routiful, 2018-07-17:
If you have any question or suggest, please ask to me any time.
Thanks :)

dhiegomaga, 2020-02-12:
I have the exact same issue as @KvalheimRacing 
So, in summary, if I want to control both servos AX (protocol version 1) and MX (that has protocol version 2), there is no way to do this **using the workbench**, unless I downgrade MX to protocol version 1?

gi0rg0sPapamichail, 2026-02-07:
If you’re running into issues using servos with mixed protocols on the same bus, one simple approach is to make a function that automatically switches the protocol before sending commands.

For example, I use something like this:

void setProtocol(uint8_t protocol) {
  if (dxl.getPortProtocolVersion() != (float)protocol) {
    dxl.setPortProtocolVersion((float)protocol);
    current_dxl_protocol = (float)protocol;
    debug_Serial_printf("Dynamixel protocol set to %.1f\n", dxl.getPortProtocolVersion());
  }
}

current_dxl_protocol is just a global float to keep track of the active protocol.

Then I wrap my servo commands in a function that takes the protocol as a parameter:

bool moveServopersistent(uint8_t id, int32_t position, uint8_t unit, int32_t speed, float protocol) {
  setProtocol(protocol);

  while (!dxl.setGoalVelocity(id, speed)) {
    debug_Serial_println("Setting speed...");
    vTaskDelay(pdMS_TO_TICKS(10));
  }

  while (!dxl.setGoalPosition(id, position, unit)) {
    debug_Serial_print("Setting position...");
    debug_Serial_print(position);
    debug_Serial_print(" ID ");
    debug_Serial_println(id);
    vTaskDelay(pdMS_TO_TICKS(10));
    printServoError(id);
  }

  return true;
}

Usage is straightforward:

moveServopersistent(17, 815, UNIT_RAW, 0, 1.0); // Protocol 1
moveServopersistent(7, 815, UNIT_RAW, 0, 2.0);  // Protocol 2

This lets you use multiple protocols on the same bus seamlessly. The main thing to remember is to set the operation mode correctly for Protocol 2 servos.