ROBOTIS-GIT/dynamixel-workbench #106: Controller Can't Open .device file

hiseholuwa, 2018-02-12:
Hi, I recently purchased a pair of the new MX-64AR dynamixels. To test them off the bat I tired the same tutorial [here](http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/CommandLine)  that I used to test my former XL430 dynamixels. I got the following error on running the launch file

`process[single_dynamixel_monitor-1]: started with pid [22938]
process[single_dynamixel_controller-2]: started with pid [22939]
[ INFO] [1518401297.463739268]: Succeeded to open the port(/dev/ttyACM1)!
[ INFO] [1518401297.466331411]: Succeeded to change the baudrate(57600)!
[ INFO] [1518401297.466454570]: Scan Dynamixel(ID: 1~253) Using Protocol 2.0

[ INFO] [1518401297.466500817]: ...wait for seconds
[ WARN] [1518401302.522304070]: Please Check USB Port authorization and
[ WARN] [1518401302.522848732]: Baudrate [ex : 57600, 115200, 1000000, 3000000]
[ERROR] [1518401302.523109159]: ...Failed to find dynamixel!

REQUIRED process [single_dynamixel_monitor-1] has died!
process has died [pid 22938, exit code -11, cmd /opt/ros/kinetic/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/hiseholuwa/.ros/log/f68eafe8-0f89-11e8-89af-94659c20af6c/single_dynamixel_monitor-1.log].
log file: /home/hiseholuwa/.ros/log/f68eafe8-0f89-11e8-89af-94659c20af6c/single_dynamixel_monitor-1*.log
Initiating shutdown!

[single_dynamixel_controller-2] killing on exit
[single_dynamixel_monitor-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

but when i run the same launch file with my XL430 connected instead, it initializes successfully and I'm able to work with the control table. On researching the e-manual I discovered that the MX-64AR shipped with protocol 1.0 and they have released a firmware update for it to protocol 2.0. I upgraded the dynamixels and tried again but I was confronted with a different error this time

`process[single_dynamixel_monitor-1]: started with pid [27273]
process[single_dynamixel_controller-2]: started with pid [27274]
[ INFO] [1518402986.258070617]: Succeeded to open the port(/dev/ttyACM1)!
[ INFO] [1518402986.266048475]: Succeeded to change the baudrate(57600)!
[ INFO] [1518402986.266133495]: Scan Dynamixel(ID: 1~253) Using Protocol 2.0

[ INFO] [1518402986.266225643]: ...wait for seconds
[ERROR] [1518402986.306155273]: Unable to open .device file : /opt/ros/kinetic/share/dynamixel_workbench_toolbox/dynamixel/models//.device

REQUIRED process [single_dynamixel_monitor-1] has died!
process has finished cleanly
log file: /home/hiseholuwa/.ros/log/f68eafe8-0f89-11e8-89af-94659c20af6c/single_dynamixel_monitor-1*.log
Initiating shutdown!

[single_dynamixel_controller-2] killing on exit
[single_dynamixel_monitor-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

I figured since the launch file utilizes protocol 2, that should fix the issue but I guess I'm wrong.  I get the same error when I try the velocity controller tutorial found [here](http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials/VelocityControl). I'll appreciate any help in resolving this. Thanks!

routiful, 2018-02-13:
Hello @hiseholuwa 

The latest version of dynamixel-workbench is updated last month. 
You can download it in master branch. I recommend to use this new version.

before you download new version, please purge the dynamixel-workbench you got

$ sudo apt-get purge ros-kinetic-dynamixel-workbench*

Thanks
Darby

hiseholuwa, 2018-02-13:
@routiful Alright, I'll try that and get back to you. Thanks!

hiseholuwa, 2018-02-13:
@routiful It works now, thanks for the help!

closed 2018-02-13 by hiseholuwa