ROBOTIS-GIT/dynamixel-workbench #112: broken install rule in dynamixel_workbench_toolbox

tfoote, 2018-02-20:
It looks to be trying to install from the wrong path here: https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/CMakeLists.txt#L60

This means the 0.2.0 release fails to build to debian packages. http://build.ros.org/view/Kbin_uxv8_uXv8/job/Kbin_uxv8_uXv8__dynamixel_workbench_toolbox__ubuntu_xenial_arm64__binary/121/console

```
00:20:40.528 Install the project...
00:20:40.538 /usr/bin/cmake -P cmake_install.cmake
00:20:40.655 -- Install configuration: "None"
00:20:40.662 -- Installing: /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/lib/pkgconfig/dynamixel_workbench_toolbox.pc
00:20:40.666 -- Installing: /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/share/dynamixel_workbench_toolbox/cmake/dynamixel_workbench_toolboxConfig.cmake
00:20:40.667 -- Installing: /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/share/dynamixel_workbench_toolbox/cmake/dynamixel_workbench_toolboxConfig-version.cmake
00:20:40.668 -- Installing: /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/share/dynamixel_workbench_toolbox/package.xml
00:20:40.670 -- Installing: /tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/lib/libdynamixel_workbench_toolbox.so
00:20:40.677 -- Set runtime path of "/tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/debian/ros-kinetic-dynamixel-workbench-toolbox/opt/ros/kinetic/lib/libdynamixel_workbench_toolbox.so" to ""
00:20:40.681 CMake Error at cmake_install.cmake:71 (file):
00:20:40.681   file INSTALL cannot find
00:20:40.681   "/tmp/binarydeb/ros-kinetic-dynamixel-workbench-toolbox-0.2.0/DynamixelWorkbench/include/dynamixel_workbench".
00:20:40.681 
```

robotpilot, 2018-02-20:
> master/dynamixel_workbench_toolbox/CMakeLists.txt#L60
> uXv8__dynamixel_workbench_toolbox__ubuntu_xenial_arm64__binary/121/console

Thank you so much.
Now, I fixed it and up PR.
Thanks,

-- 
//------------------------------------------------------------------------------
Yoonseok Pyo, Ph.D.,
Senior Research Engineer / Open Source Team / R&D Dept.
ROBOTIS CO., LTD. (http://www.robotis.com)
#1505 Ace High-End Tower NO. 3, Gasan-dong 145,
GasanDigital 1-ro, GeumCheon-gu, Seoul, Korea 153-787

Follow ROBOTIS on Twitter <https://twitter.com/robotis>, Facebook
<https://www.facebook.com/robotis.company>, Linkedin
<https://www.linkedin.com/company/robotis>, Github
<https://github.com/ROBOTIS-GIT>
Personal Website (http://www.robotpilot.net)
//------------------------------------------------------------------------------

tfoote, 2018-02-22:
It looks like there's a similar issue still failing to build with the new release: http://build.ros.org/view/Kbin_uxv8_uXv8/job/Kbin_uX64__dynamixel_workbench_toolbox__ubuntu_xenial_amd64__binary/128/console

tfoote, 2018-02-22:
It's not a problem. The notifications are mostly for your benefit. I'll go ahead with the sync without waiting.

robotpilot, 2018-02-28:
Solved! ref issues: https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/117

closed 2018-02-28 by robotpilot