ROBOTIS-GIT/dynamixel-workbench #115: dynamixel_item PROInfo Max&Min radian Position Problem

martian491, 2018-02-26:
Hi

According to the following two lines:

These two values belong to H54-100-S500-R & H54-200-S500-R

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/1e767b9a1d429d1f4fbf31f43511aa08a9114fc9/dynamixel_workbench_toolbox/src/dynamixel_workbench/dynamixel_item.cpp#L1081

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/1e767b9a1d429d1f4fbf31f43511aa08a9114fc9/dynamixel_workbench_toolbox/src/dynamixel_workbench/dynamixel_item.cpp#L1083

But in H42-20-S300-R,
 max radian position is 151875
min  radian position is -151875 

According to the [Control table](http://support.robotis.com/en/product/actuator/dynamixel_pro/dynamixelpro/control_table.htm)

thx

routiful, 2018-02-27:
Hello :)

Thank you for your inquiry.
I will be update it as soon as possible.

Thanks
Darby

routiful, 2018-03-08:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-08 by routiful