ROBOTIS-GIT/dynamixel-workbench #127: Baudrate [ex : 9600, 57600, 115200, 1000000, 2000000] ...Failed to find dynamixel!

cjsend56, 2018-03-08:
Hi, 
I just downloaded packages and tried to follow tutorial but I got error message 

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /ping: False
 * /ping_id: 1
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /
    single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
    single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

ROS_MASTER_URI=http://localhost:11311

process[single_dynamixel_monitor-1]: started with pid [4178]
process[single_dynamixel_controller-2]: started with pid [4179]
Please Check USB Port authorization and
Baudrate [ex : 9600, 57600, 115200, 1000000, 2000000]
...Failed to find dynamixel!
================================================================================REQUIRED process [single_dynamixel_monitor-1] has died!
process has died [pid 4178, exit code 1, cmd /home/hoon/catkin_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/hoon/.ros/log/794a0d96-231d-11e8-944e-180373ba866f/single_dynamixel_monitor-1.log].
log file: /home/hoon/.ros/log/794a0d96-231d-11e8-944e-180373ba866f/single_dynamixel_monitor-1*.log
Initiating shutdown!
================================================================================
[single_dynamixel_controller-2] killing on exit
[single_dynamixel_monitor-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

How can I fix it?

routiful, 2018-03-09:
Hi :)

You can change baudrate number in launch file.
Please check baudrate supported by your dynamixel in [wiki](http://emanual.robotis.com/)

Thanks
Darby

cjsend56, 2018-03-19:
Hi Darby,

It seems like I am having trouble with LN-101, because it works with my USB2Dynamixel 
Do I need to make some changes to for LN-101?

routiful, 2018-03-19:
Have you succeed to connect Dynamixel to Dynamixel-workbench by using USB2Dynamixel?
Would you explain me how to connect LN-101 to Dynamixel?

cjsend56, 2018-03-20:
I used CM-700 controller to connect. 
Yes, it worked with USB2Dynamixel through TTL connection, but LN-101 connection does not work..

routiful, 2018-03-21:
If you want to use CM-700 with LN-101, you need to firmware to pass packet for controlling Dynamixel.

OpenCR has [example code](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino)