ROBOTIS-GIT/dynamixel-workbench #128: Failed to find dynamixel! with dynamixel XM430_W210

Angit16, 2018-03-13:
I am using dynamixel XM430-W210. I recently changed its baudrate from 1Mbps to 3 Mbps using the dynamixel_manager_gui. Then, I when I went to test it, it was showing port authorization error. Then, I used Firmware recovery of R+ manager and reset it to 57600bps. Now, when I used the launch file single_manager.launch within the  dynamixel_workbench_single_manager package, I found following errors. I am not sure, why it is happening. I again did firmware recovery but it gives same result. I am confused.

```
NODES
  /
    single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
    single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

auto-starting new master
process[master]: started with pid [29790]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2a97f42e-26ec-11e8-ac4c-28f10e2a37e5
process[rosout-1]: started with pid [29803]
started core service [/rosout]
process[single_dynamixel_monitor-2]: started with pid [29807]
process[single_dynamixel_controller-3]: started with pid [29813]
[ INFO] [1520965641.001409086]: Succeeded to open the port(/dev/ttyUSB0)!
[ INFO] [1520965641.002667441]: Succeeded to change the baudrate(57600)!
[ INFO] [1520965641.002854747]: Scan Dynamixel(ID: 1~253) Using Protocol 1.0

[ INFO] [1520965641.002903324]: ...wait for seconds
[ WARN] [1520965641.174536507]: Please Check USB Port authorization and
[ WARN] [1520965641.174578832]: Baudrate [ex : 57600, 115200, 1000000, 3000000]
[ERROR] [1520965641.174604571]: ...Failed to find dynamixel!
[ INFO] [1520965641.174623935]: dynamixel_workbench_single_manager : Init Success!
================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
process has died [pid 29807, exit code -11, cmd /home/anindya/catkin_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/anindya/.ros/log/2a97f42e-26ec-11e8-ac4c-28f10e2a37e5/single_dynamixel_monitor-2.log].
log file: /home/anindya/.ros/log/2a97f42e-26ec-11e8-ac4c-28f10e2a37e5/single_dynamixel_monitor-2*.log
Initiating shutdown!
================================================================================
[single_dynamixel_controller-3] killing on exit
[single_dynamixel_monitor-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
anindya@anindya-Latitude-E7270:~/catkin_ws$ 

```

Angit16, 2018-03-14:
ok, it works now.

closed 2018-03-14 by Angit16

msaufyrohmad, 2018-05-24:
how do u make it works? i am using xm540-w270. the same error occur

routiful, 2018-05-25:
Hi :) @msaufyrohmad  

This issue had been closed. Please open another issue with detail include used commands and error messages

Thanks

msaufyrohmad, 2018-05-25:
thank you. can i send the problem try email with u?

saufy

routiful, 2018-05-25:
Because every issue can help other users, I recommend you would open new issue.

msaufyrohmad, 2018-05-25:
ok thnks