ROBOTIS-GIT/dynamixel-workbench #129: dynamixel XM-430W210 low publishing frequency

Angit16, 2018-03-14:
I am running dynamixel XM-430_W210 on a i5 (4 core (2.5GHz)) laptop. I checked that if I run the dynamixel_workbench_tutorial single_manager.launch, it usually offers publishing rate of 1.2-1.5Hz. Then, I looked into https://github.com/jysung/dynamixel-workbench/commit/26fc2a2c7d507aeee66a80969713c28ae9c85757 and modified as mentioned and I was getting 

```
    // ************************************
    #include <ros/ros.h>
    #include <std_msgs/Float64.h>
    #include <laser_assembler/AssembleScans.h>
    #include <std_msgs/Time.h>
    #include <std_msgs/Empty.h>
    #include <iostream>
    #include <inttypes.h>
    #include <ros/time.h>
    #include "ros/ros.h"
    // #include <cmath>
    #include "dynamixel_workbench_msgs/DynamixelCommand.h"
    #include "dynamixel_workbench_single_manager/single_dynamixel_monitor.h"

    using namespace single_dynamixel_monitor;
    // using namespace std;

    #define DXL_MINIMUM_POSITION_VALUE      108                // Dynamixel will rotate between this value approx. (87.936 deg)
    #define DXL_MAXIMUM_POSITION_VALUE      2650                // and this value (note that the Dynamixel would not move when
    #define DXL_MOVING_STATUS_THRESHOLD     5                 // Dynamixel moving status threshold
    #define SPEED                           40                // set speed count

    uint32_t dxl_goal_position[2] = {DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE};         // Goal position
    uint32_t count=0; float error; int max_angle; int min_angle; float pause_time;

    SingleDynamixelMonitor::SingleDynamixelMonitor()
    {
        // Check Dynamixel Ping or Scan (default : Scan (1~253))
        use_ping_ = node_handle_.param<bool>("ping", false);
        ping_id_  = node_handle_.param<int>("ping_id", 1);

        // Load Paramameter For Connection
        dynamixel_info_ = new dynamixel_driver::DynamixelInfo;

        dynamixel_info_->lode_info.device_name      = node_handle_.param<std::string>("device_name", "/dev/ttyUSB0");
        dynamixel_info_->lode_info.baud_rate        = node_handle_.param<int>("baud_rate", 57600);
        dynamixel_info_->lode_info.protocol_version = node_handle_.param<float>("protocol_version", 1.0);

        dynamixel_driver_ = new dynamixel_driver::DynamixelDriver(dynamixel_info_->lode_info.device_name,
                                                                  dynamixel_info_->lode_info.baud_rate,
                                                                  dynamixel_info_->lode_info.protocol_version);

        if(!use_ping_)
        {
            // Get Connected Single Dynamixel State
            if (dynamixel_info_->lode_info.protocol_version == 1.0)
                ROS_INFO("Scan Dynamixel(ID: 1~253) Using Protocol 1.0\n");
            else if (dynamixel_info_->lode_info.protocol_version == 2.0)
                ROS_INFO("Scan Dynamixel(ID: 1~253) Using Protocol 2.0\n");

            if (dynamixel_driver_->scan())
            {
                ROS_INFO("...Succeeded to find dynamixel\n");
                ROS_INFO("[ID] %u, [Model Name] %s, [BAUD RATE] %d",
                         dynamixel_driver_->dynamixel_->id_, dynamixel_driver_->dynamixel_->model_name_.c_str(), dynamixel_info_->lode_info.baud_rate);
            }
            else
            {
                ROS_WARN("Please Check USB Port authorization and");
                ROS_WARN("Baudrate [ex : 57600, 115200, 1000000, 3000000]");
                ROS_ERROR("...Failed to find dynamixel!");
                // shutdownSingleDynamixelMonitor();
            }
        }
        else
        {
            // Ping Connected Single Dynamixel State
[... 334 lines cut ...]
{      xm_state.present_velocity = read_value;
    // std::cout <<"present_velocity" << read_value <<std::endl;
    }    else if ("present_position" == dynamixel->item_->item_name)
{      xm_state.present_position = read_value;
    SingleDynamixelMonitor::VAL = read_value;
    std::cout <<"present_position" << read_value <<std::endl;
    }    else if ("velocity_trajectory" == dynamixel->item_->item_name)
{      xm_state.velocity_trajectory = read_value;
    // std::cout <<"velocity_trajectory" << read_value <<std::endl;
    }    else if ("position_trajectory" == dynamixel->item_->item_name)
{      xm_state.position_trajectory = read_value;
    // std::cout <<"position_trajectory" << read_value <<std::endl;
    }    else if ("present_input_voltage" == dynamixel->item_->item_name)
{      xm_state.present_input_voltage = read_value;
    // std::cout <<"present_input_voltage" << read_value <<std::endl;
    }    else if ("present_temperature" == dynamixel->item_->item_name)
{      xm_state.present_temperature = read_value;
    // std::cout <<"present_temperature" << read_value <<std::endl;
  }  
}

  dynamixel_status_pub_.publish(xm_state);

}
```

routiful, 2018-03-14:
Hi

Purpose of **single_monitor package** is to check your Dynamixel status. 
I highly recommend if you want to test position control, you could use **controllers package**

Thanks
Darby

Angit16, 2018-03-14:
That's fine. Could you please have a look, into the code? I don't understand why there should such huge change in publishing frequency?

routiful, 2018-03-15:
I don't think so that the differences are caused by code. 
But i need to update package to improve frequency as soon as possible.
Actually, this package is updated few weeks ago. I recommend to use it.

Thanks

Angit16, 2018-03-17:
I solved this problem. Actually, the USB latency timer in my ubuntu laptop was by default 16 and the same in desktop was 1. I reduced it from 16 to 1 and now I for the above code, I get 523Hz approximately.

routiful, 2018-03-18:
Would you write down that how to change USB latency timer in ubuntu?
It will be help to other people who have faced same problem.

Thanks
Darby

Angit16, 2018-03-21:
I was away for couple of days...

 FTDI serial to USB converter chip is used in dynamixel to USB converter which uses latency timer. By default the delay could be as high as 16 msec. We can check it in linux terminal by
`cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer`
To change it to a smaller value following steps may work
```
 sudo usermod -aG dialout $USER
 echo 4 | sudo tee /sys/bus/usb-serial/devices/ttyUSB0/latency_timer
```
Then check it again by
` cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer 4`

routiful, 2018-03-21:
Thank you very much !

closed 2018-04-08 by routiful

jackwilkinson255, 2020-07-22:
Does anyone know how to reduce the usb latency in ubuntu 18.04?