ROBOTIS-GIT/dynamixel-workbench #133: rostopic can't return the motors' state?

ynfei, 2018-03-28:
i use the dynamixel-workbench position control, when i want to return the motors' state ,i use this rostopic echo /dynamixel_state ,but i get this:
dynamixel_state: 
  - 
    model_name: "MX-106"
    id: 1
    torque_enable: 0
    goal_current: 0
    goal_velocity: 0
    goal_position: 0
    present_current: 0
    present_velocity: 0
    present_position: 0
    moving: 0
  - 
    model_name: "MX-64"
    id: 2
    torque_enable: 0
    goal_current: 0
    goal_velocity: 0
    goal_position: 0
    present_current: 0
    present_velocity: 0
    present_position: 0
    moving: 0
  - 
    model_name: "MX-28"
    id: 3
    torque_enable: 0
    goal_current: 0
    goal_velocity: 0
    goal_position: 0
    present_current: 0
    present_velocity: 0
    present_position: 0
    moving: 0
how could i fix this problem ,thank you!

routiful, 2018-03-29:
Hi :)

Did you check baud rate of Dynamixel?

routiful, 2018-04-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-04-16 by routiful