ROBOTIS-GIT/dynamixel-workbench #134: Position controller package issue

cjsend56, 2018-04-02:
Hi, 

I am trying to connect 16 motors(4 MX-106, 2 MX-64, 4 MX-28, 6 AX-12A) at the same time, but i get error if I connect all.
Is it because I use AX and MX motors together?

Below is the what I get when I try

started roslaunch server http://hoon-OptiPlex-790:37115/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /scan_range: 200

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

ROS_MASTER_URI=http://localhost:11311

process[position_control-1]: started with pid [12660]
================================================================================REQUIRED process [position_control-1] has died!
process has died [pid 12660, exit code -11, cmd /home/hoon/catkin_ws/devel/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/hoon/.ros/log/0e6e915e-36a5-11e8-aa54-180373ba866f/position_control-1.log].
log file: /home/hoon/.ros/log/0e6e915e-36a5-11e8-aa54-180373ba866f/position_control-1*.log
Initiating shutdown!
================================================================================
[position_control-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

routiful, 2018-04-03:
Hi :)

Would you please double check baud rate and ID?

cjsend56, 2018-04-03:
I double checked id and baud rate, and still does not work..

When I tested with other package(dynamixel_controllers), all motors worked

routiful, 2018-04-04:
Is it not working when you try same series? 
I checked it again but it is working well.

Sorry for inconvenience.

routiful, 2018-04-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-04-16 by routiful