ROBOTIS-GIT/dynamixel-workbench #140: Limit velocity in position control mode

djmira92, 2018-04-16:
How can I limit the velocity of the movelment in position control mode?

I tried to set the Velocity_Limit but that only works in velocity mode.

routiful, 2018-04-16:
Hi :)

You can make profile to control Dynamixel by [Profile Velocity and Profile Acceleration](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#profile-velocity112).

This function would be help you

robotpilot, 2018-04-23:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-04-23 by robotpilot

gpicardi, 2018-12-07:
Hi,
I am using XM430-w350 in position control mode with a rectangular profile (i.e. acceleration profile = 0 as by default and velocity profile != 0). I set Velocity Limit in the RAM to 1023.

It seems to me that Profile velocity values above 330  do not produce a higher velocity, however I cannot find any reference to that in the eManual nor in any issue.

Is the behaviour normal or am I missing somenthing?

Many thanks in advance,
Giacomo

madlink306, 2019-08-09:
Hi, I don't know where to post this. So maybe here. Sorry if it's not the right place, and please point me to where I can ask this kind of question.

So, anyway, I'm trying to learn to control dynamixel motr. And I know we can use velocity and acceleration profile to create trapezoidal profile.
1. Which value should I set for velocity profile?
For example I read in the datasheet that at 12V the velocity is 30 rev/min. As I understand that I should set the value of velocity profile to 132 (30 divided to 0.229 rev/min = 132). Is that correct?

2. What about the acceleration profile? I see nothing mention in the datasheet so I don't know what is the max acceleration for the motor. The datasheet say that, acceleration profile value will not exceed 50% of velocity. Does that mean I should set the acceleration value to 61 (50% of 132)?

3. Time calculation for trapezoidal profile is t1 = 64 * (Velocity profile/Acceleration profile), t2 = 64 * delta_position / Velocity profile.
What is the units and values using in here? Is that the real velocity value (30 rev/min) or the integer value set before(132), same with acceleration?