ROBOTIS-GIT/dynamixel-workbench #141: Connect MoveIt! with real robot (hardware)

robotpilot, 2018-04-17:
https://answers.ros.org/question/288940/connect-moveit-with-real-robot-hardware/

znbrito, 2018-04-29:
Hello Pyo,

what does this closed issue mean?

Best regards,
José Brito

robotpilot, 2018-04-29:
Your issue is very distributed like 'answers.ros.org', 'GitHub issues' and 'emails to three developers'. Currently, your issue is mainly mailed, so I have removed this issue. If you agree, we hope to keep only GitHub issues, not other channels.

znbrito, 2018-04-29:
???

robotpilot, 2018-04-29:
You send e-mail us that the hardware connection was successful like the title of this issue. This issue is therefore closed. If you have new issues, register new issues. Thanks.

znbrito, 2018-04-29:
Oh, I see! I may post eveything on that issue post from now on! :) Please
keep it open, I will post my advances there tomorrow (its midnight here).

Best regards,
José Brito

znbrito, 2018-04-30:
Hello all,

I have been trying to make this work for a long time and I noticed some problems when I adapted the "open_manipulator" packages for the Robotis Manipulator-H. The main problem is that the 5th and 6th joint values are not being correctly read by the the "open_manipulator_dynamixel_ctrl" package. In order to overcome that, I tried to use the package "manipulator_h_manager" to serve as an interface between the PC and the robot. The launch file in this package launches a node that sends the joint values of the manipulator through a sensor_msgs/JointState message published on a topic. He also subscribes this kind of messages on another topic, and I thought I could send the MoveIt! trajectories through this topic. The problem is that although the MoveIt! messages with the trajectories were being published in the topic, the manager did seem to ignore them. I then tried to use the package "manipulator_h_gui" to interface with the manager and make single joint movements and I understood that the type of messages being subscribed by the manager that made him move were completely different from a sensor_msgs/JointState, so I discarded the option of having the "manipulator_h_manager" package serving as an interface for the robot and the PC.

Fortunately, a guy in my lab a guy had previously worked with the Robotis Manipulator-H and had developed a package that interfaces with the robot. This package publishes the joint values of the robot in a sensor_msgs/JointState message and also subscribes goal joint values in this same topic, so this is exactly what I wanted. As this node reads all joint values with now problem, I simply had to change the names of the topics being published and subscribed by this node and now everything works fine, I can plan with the MoveIt! GUI in RViz and the planned trajectories are followed perfectly by the robot. 

The main problem now is: how can I code trajectories like what is made in this tutorial: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html ? Because of course I want to code trajectories and not simply being able to execute them via the MoveIt! GUI in RViz. I don't see anything developed for the open manipulator robot, did you worked on this?

Note that you guys have the "fake_execution" arg set to true in the launch file for MoveIt! in the open manipulator pacakges: https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_moveit/launch/open_manipulator_demo.launch . Whenever I set this value to false, I get a lot of problems, as you can see in my post in ROS answers: https://answers.ros.org/question/289574/moveit-error-cannot-execute-trajectories-in-real-robot/

Summing up, my main problem now is getting the robot to execute coded trajectories, any idea on how to do this? 

Best regards,
José Brito

routiful, 2018-05-02:
Hi again :)

You can find a code to use `move_group` in OpenManipulator package

https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_position_ctrl/src/position_controller.cpp#L249

It is simple communication to use ROS message.

1. Send goal pose of end-effector
2. Start plan
3. If plan was succeed, get planned path
4. Publish goal joint position

Thanks.

routiful, 2018-05-03:
Yes, i know your problem. Unfortunately, i don't know how to solve yours. Because i faced that same issue.
So i have tried different way that get a planned path and publish it to goal joint position. That's what i said before comment.

Thanks.

znbrito, 2018-05-03:
Hello,

yes, that's what I thought you were doing. I understood that later today because I noticed, by observing the behavior of the manipulator, that it doesn't quite move as If he was following a trajectory, he simply moves his joints one by one, from joint number 1 to joint number 6, till it reaches the desired final position for each joint. This means that you get all the points of the trajectory and only use the last one (final position) and send it as a sensor_msgs/JointState to the manipulator (as you said your last 2 comments), right? 

If this is right, I believe that right now you guys can't make your manipulator (and mine also) to follow a trajectory, am I correct? Are you working on this?

Best regards,
José Brito

routiful, 2018-05-09:
Hi :)

> This means that you get all the points of the trajectory and only use the last one (final position) and send it as a sensor_msgs/JointState to the manipulator (as you said your last 2 comments), right?

The answer is NO. If I sent goal_pose(x,y,z,roll,pitch,yaw) to MoveIt!, a path are calculated. After it succeeded, you can get a each goal position of joints by planned_path topic. You can pass this data to goal joint topic to control your manipulator.

Thanks

znbrito, 2018-05-14:
Hello,

so you are saying that right now you don't have an implemented function that tracks and makes it possible for the manipulator to follow a path planned and that I should do it?

Thanks!

routiful, 2018-05-15:
Your right. We have a plan to implemented it but not specific in detailed. If you could find more simple method to send planned path to Dynamixel, Please let me know how to do it and requeset PR. 

Thanks :)

znbrito, 2018-05-18:
Ok, thanks a lot! Right now I don't have time to look at that specific problem but I hope that you guys can solve it soon because that is an important condition for MoveIt! to be fully working with your robots. 

Best of lucks!
José Brito