ROBOTIS-GIT/dynamixel-workbench #147: [FATAL] [1527235095.902048]: pan_tilt_port: No motors found. - fail to connect to XM540! help me

msaufyrohmad, 2018-05-25:
hi all,
iam new to ros. my opencr and dynamixel XM540-W270-R is working fine on arduino using dynamixel_workbench.h library. 

however, using when using ros. i failed to follow both tutorial by dynamixel workbench and dynamixel controller.

the error is: [FATAL] [1527235095.902048]: pan_tilt_port: No motors found.

i am using /dev/ttyACM0 and already change the .launch file to that port with 57600 bout rate. 

using another tutorial also failed to connect to the motor. 

Please Check USB Port authorization and
Baudrate [ex : 9600, 57600, 115200, 1000000, 2000000]
...Failed to find dynamixel!

can anyone give me some light please. i also already make soft link between /dev/ttyACM0 to /dev/ttyUSB0. but the result is still same. 

help me please

saufy

routiful, 2018-05-28:
Hello :)

Thank you for inquiry about Dynamixel-Workbench.

If you want to launch it, you have to connect Dynamixel to [USB2Dynamixel](http://emanual.robotis.com/docs/en/parts/interface/usb2dynamixel/) or [U2D2](http://www.robotis-shop-en.com/?act=shop_en.goods_view&GS=3288&keyword=u2d2). Then you can watch a port is opened for example `/dev/ttyUSB0`
Or else, you have to upload [usb_to_dxl](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino) to OpenCR(Not included in OpenCm9.04). Then you can watch a port is opened for example `/dev/ttyACM0`

Thanks

msaufyrohmad, 2018-05-28:
thank  you very much! i already run it sucessfully. yes we need usb_to_dxl first (not mention in the emanual).

closed 2018-05-28 by msaufyrohmad

routiful, 2018-05-28:
Nice! I am going to add this process in WIKI.

Thanks :)