ROBOTIS-GIT/dynamixel-workbench #148: dynamixel_workbench - torque_control.launch REQUIRED process [torque_control-1] has died!

squirel2000, 2018-05-25:
While I executed the torque_control.launch in dynamixel_workbench, the two motors will perform the current control correctly, but the roslaunch will die with exit code -11, after several seconds or minutes. The log is as follow,

PARAMETERS
 * /baud_rate: 1000000
 * /d_gain: 2e-05
 * /device_name: /dev/ttyUSB0
 * /p_gain: 0.003
 * /pan_id: 6
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /tilt_id: 7

NODES
  /
    torque_control (dynamixel_workbench_controllers/torque_control)

ROS_MASTER_URI=http://192.168.234.137:11311

process[torque_control-1]: started with pid [29619]
-----------------------------------------------------------------------
  dynamixel_workbench controller; torque control example               
-----------------------------------------------------------------------

MODEL   : XM430-W350
ID      : 6

MODEL   : XM430-W350
ID      : 7

-----------------------------------------------------------------------
================================================================================REQUIRED process [torque_control-1] has died!
process has died [pid 29619, exit code -11, cmd /home/asus/catkin_ws/devel/lib/zenbo_dynamixel_workbench_controllers/torque_control __name:=torque_control __log:=/home/asus/.ros/log/3511a6ae-6001-11e8-86f4-000c295ac8c7/torque_control-1.log].
log file: /home/asus/.ros/log/3511a6ae-6001-11e8-86f4-000c295ac8c7/torque_control-1*.log
Initiating shutdown!
================================================================================

Any comments are appreciated!

routiful, 2018-05-28:
Hi :)

Would you check frequency of topic to use below command?

```
$ rostopic hz /dynamixel_state
```
I assumed that might be no return data from Dynamixel.

Or try again with more higher baudrate as 3000000

Thanks.

squirel2000, 2018-05-29:
Hi routiful,

Thanks a lot for your reply. I checked the frequency:
average rate: 25.643
	min: 0.033s max: 0.104s std dev: 0.01098s window: 282

By the way, I also changed the baud_rate as 3M as follow,
PARAMETERS
 * /baud_rate: 3000000
 * /d_gain: 2e-05
 * /device_name: /dev/ttyUSB0
 * /p_gain: 0.003
 * /pan_id: 6
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /tilt_id: 7

However, the same issue and error message, I got in 1 minute after executing the torque_controller.launch, is pasted as follow for your reference:
================================================================================REQUIRED process [torque_control-2] has died!
process has died [pid 21953, exit code -11, cmd /home/asus/zenbo_ws/devel/lib/dynamixel_workbench_controllers/torque_control __name:=torque_control __log:=/home/asus/.ros/log/e59e07e2-62f4-11e8-b645-000c295ac8c7/torque_control-2.log].
log file: /home/asus/.ros/log/e59e07e2-62f4-11e8-b645-000c295ac8c7/torque_control-2*.log
Initiating shutdown!
================================================================================

Thanks

routiful, 2018-05-29:
Hi :) 

Would you try to update firmware by using [RoboPlus Manager 2.0](http://emanual.robotis.com/docs/en/software/rplus2/manager/)?

My XM430-W350 is set by latest version and its working after 1 minute.

However, if it occur same problem after updating it, please show me state of topic (/dynamixel_state) just before it crashed 

Thanks

squirel2000, 2018-05-30:
Hi routiful,

The firmware of XM430-W350 is 41 shown in R+ manager.
This issue does not always happen in one minute. It may happen in one minute, but sometimes it takes several minutes. For example, I have this issue today after about 6 minutes, but this issue happened within 1 minute.  The state of the topic before it crashed is listed as follows:
------------------------------------
average rate: 34.494
	min: 0.023s max: 0.108s std dev: 0.00500s window: 13111
average rate: 34.483
	min: 0.023s max: 0.108s std dev: 0.00505s window: 13123
no new messages
no new messages
------------------------------------

Thanks

routiful, 2018-05-31:
Sorry for inconvenience. 

My environments is set by two XM430-W350 is connected with [U2D2](http://www.robotis-shop-en.com/?act=shop_en.goods_view&GS=3288&keyword=u2d2) in Kinetic (OS is Linux Mint 18.1)
Would you tell me your environments in detail?

Thanks.

squirel2000, 2018-05-31:
Hi routiful,

My operating system is kinetic ROS in 64-bit ubuntu 16.04 LTS, which is executed in virtual machine.  We have the same hardware, XM430-W350 connected via U2D2. 

Thanks,

routiful, 2018-06-04:
Every thing is alright but we don't recommend to install Ubuntu in virtual machine.
If you have a chance to get resources, please install Ubuntu in isolate PC or dual booting it.

So, we tested that example during all day, but we didn't get any problems. 

Thanks

routiful, 2018-06-07:
Good. If you will get same error in pure Ubuntu system, please you would re-open this issue.

Thanks

closed 2018-06-07 by routiful