ROBOTIS-GIT/dynamixel-workbench #150: Dynamixel Workbench fo ROS Melodic Morenia

AchmadFathoni, 2018-05-29:
There still no dynamixel_workbench metapackage for ROS Melodic Morenia in official ROS repository but there is already branch for it in github. How to properly install dynamixel_workbench for ROS Melodic Morenia?

robotpilot, 2018-05-29:
Hi, @AchmadFathoni,

ROS Kinetic has more than 2,000 packages registered, while ROS Melodic is currently registered with just over 400 packages.
http://repositories.ros.org/status_page/ros_melodic_default.html

We are currently working on a dynamixel_workbench package that runs on the ROS Melodic version. We are currently waiting for some dependency packages to be resolved.

I will let you know when this issue is resolved and the ROS Melodic version is released.

robotpilot, 2018-06-04:
Hi @AchmadFathoni,
Today, we have registered binary packages for the ROS Melodic version. It is now waiting for registration.
http://repositories.ros.org/status_page/ros_melodic_default.html?q=dynamixel
You can download it at the next sync via `sudo apt-get install ros-melodic-dynamixel-workbench`.