ROBOTIS-GIT/dynamixel-workbench #157: failed to call service /wheel_command    (ax-12a)

JanPieterDeRie, 2018-06-07:
Hi, I am working with the AX-12A and just trying to make them spin, so Im following the velocity tutorial. But when I try to use the wheel operator it says:
[ INFO] [1528376268.297972156]: w : Forward
[ INFO] [1528376268.297998120]: x : Backward
[ INFO] [1528376268.298019079]: a : Left
[ INFO] [1528376268.298026139]: d : Right
[ INFO] [1528376268.298032941]: s : STOPS

[ INFO] [1528376268.298040660]: ESC : exit
w[ERROR] [1528376272.615343864]: Failed to call service /wheel_command

Does anyone have any helpful information?
Thanks in regards

JanPieterDeRie, 2018-06-07:
Already solved it, the problem was that i was using one servo instead of 2.
This meant I had to change syncWrite to itemWrite for the goal_velocity

routiful, 2018-06-07:
Nice, if you have any problem, please feel free to open any time :)

JanPieterDeRie, 2018-06-08:
Hi, so ive got the second servo now but now im back to the same problem. Im still working with the ax-12a and the only thing I changed to the code is;
dxl_wb_->itemWrite(dxl_id_[1], "Drive_Mode", 1); to
//dxl_wb_->itemWrite(dxl_id_[1], "Drive_Mode", 1);

and 

  goal_velocity[1] = dxl_wb_->convertVelocity2Value(dxl_id_[1], (-1) * req.right_vel); to
  goal_velocity[1] = dxl_wb_->convertVelocity2Value(dxl_id_[1], req.right_vel);

I did this because it killed the launch and I didnt need it.

But now it seems that the program is stuck at;
bool ret = dxl_wb_->syncWrite("Goal_Velocity", goal_velocity);

With ROS_INFO I found out that the program goes to the dynamixel_workbench.cpp and then send to driver_.syncWrite. But when I put comments in
bool DynamixelDriver::syncWrite(const char *item_name, int32_t *data)
they never get shown. So that means the syncWrite command doesnt interact with its driver.
Its weird because itemWrite works just fine.

It could be that I build the package wrong but maby someone else is having the same problem.

Hope anyone can help

routiful, 2018-06-11:
Hi again :)

Sorry for late reply and thank you so much to share your comments. 
It will be nice information when other users face similar environments.

First. Because AX, RX, EX series don't have proper encoder to calculate velocity([Ref](http://emanual.robotis.com/docs/en/dxl/ax/ax-12w/#moving-speed-32)) , example of velocity control doesn't support AX, RX, EX series of Dynamixel. 
So I added message like this

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/velocity_control.cpp#L80

But maybe you can't find it. 

Second, nevertheless, if you want to use AX motor to wheel mode, you are supposed to delete below line

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/velocity_control.cpp#L56

and change control table item from `Goal_Velocity` to `Moving_Speed`

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/velocity_control.cpp#L61

and change function from syncWrite to goalSpeed
for example 

```
dxl_wb_->goalSpeed(dxl_id[0], goal_velocity[0]);
dxl_wb_->goalSpeed(dxl_id[1], goal_velocity[1]);
```

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/velocity_control.cpp#L142

However, as I mentioned that AX motor isn't controlled by velocity.

Thanks

JanPieterDeRie, 2018-06-11:
Thank you for the explanation! 

So is itemWrite and syncWrite only meant for the velocity control?
And goalSpeed also works for daisy chaining right, just to be sure.

I know this is a bit sketchy but the max rpm of ax-12a is 59rpm. So doesnt that meant that 512 = 50% = 29/30 rpm. Or is the interval of the max rpm(50 - 59) too big to work with.

Thanks in regards!

routiful, 2018-06-11:
No, itemWrite and syncWrite are functions of Dynamixel Protocols.
It used to send packets to any control table items.

You can check it more detail in below link

http://emanual.robotis.com/docs/en/dxl/protocol1/#instruction

Thanks :)

closed 2018-06-20 by robotpilot