ROBOTIS-GIT/dynamixel-workbench #158: After killing Dynamixel Workbench Single manager, all settings we have changed again set to factory...How to set these Dynamixel command changes reamin forever until I change them again?

abhayvar, 2018-06-17:


abhayvar, 2018-06-17:
Moreover the reboot option of GUI is disable to click,

routiful, 2018-06-17:
Please re-open your issue with proper description. You can find other open or closed issue.

routiful, 2018-06-18:
Hi
Dynamixel has two type of storage as EEPROM and RAM. You can check it what is a difference between them in [e-manual](http://emanual.robotis.com/docs/en/dxl/mx/mx-28/#area-eeprom-ram)

So, you are supposed to write value on RAM address after power on. 

Moreover, In Single manager, reboot option is blocked when Dynamixel using protocol 1.0 is connected

And next time Please describe what is your problem in comment(NOT in Title)

abhayvar, 2018-06-18:
Sorry for asking the question in title...can you please tell me from where to change thes ram values of pid if I'm using Ubuntu?

routiful, 2018-06-18:
That's fine. I attached example for setup PID value.
[itemWrite](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L340) function helps set value to address each Dynamixel.

```
dxl_wb_->itemWrite(DXL_ID, "P_Gain", P_Value);
dxl_wb_->itemWrite(DXL_ID, "I_Gain", I_Value);
dxl_wb_->itemWrite(DXL_ID, "D_Gain", D_Value);
```

You will add above code in setup function.

Thanks

abhayvar, 2018-06-18:
I'm sorry but I don't get it
I have just cloned dynamixel_workbench, now which packages do I need to make and which files do I need to run to set them?

routiful, 2018-06-19:
I recommend [position_control](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/position_control.cpp#L49) example. 
You can add above code on it.

abhayvar, 2018-06-19:
Thanks it is working...but position control is working for only 2 servos, we are using 10 servos currently and in launch file I have added other 8 roservice and rosparams
It run but give output of only two initial servos

abhayvar, 2018-06-19:
Thanks for your great help....now i am able to do it, but still I found that when I turned off power these pid parameters again set to 0, Is there no way to fix them?

abhayvar, 2018-06-19:
Hey! It worked...Now Just one thing more, actually I'm trying to move a biped, so how to give angles to more than one servo simultaneously  and make a script for that...In position control example I can only give angles one by one.
Thanks

routiful, 2018-06-19:
Great! You can use `goal_dynamixel_position` topic to move Dynamixel simultaneoulsy.
This topic is configured by [sensor_msgs::JointState](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/JointState.html)

abhayvar, 2018-06-20:
What is secs and nsecs in that msg?

closed 2018-06-26 by robotpilot