ROBOTIS-GIT/dynamixel-workbench #160: C header conflict on dynamixel sdk.

msaufyrohmad, 2018-06-28:
ISSUE TEMPLATE ver. 0.1.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/)

1. Which distro of ROS have you installed? (ex, indigo, kinetic, melodic)
kinetik

2. Which OS have you installed? (ex, Ubuntu 16.04 LTS (Xenial Xerus), Linux Mint 18.x, ...)
ubuntun

3. Which Dynamixel have you used? (ex, MX-28, XM430-W350, ...)
XM540-W270-R
 
4. Write down the commands you used in order
i am writing my own package to take msg in subsciber using C. but i think the default header file on dynamixel_sdk is pointing only to cpp header files.
 
5. Copy and Paste your error message on terminal
[ 48%] Built target beginner_tutorials_generate_messages
CMakeFiles/sendi_subscriber.dir/src/sendi_subscriber.cpp.o: In function `move_position(int)':
sendi_subscriber.cpp:(.text+0x605): undefined reference to `portHandler(char const*)'
sendi_subscriber.cpp:(.text+0x60d): undefined reference to `packetHandler()'
sendi_subscriber.cpp:(.text+0x63e): undefined reference to `openPort(int)'
sendi_subscriber.cpp:(.text+0x660): undefined reference to `setBaudRate(int, int)'
sendi_subscriber.cpp:(.text+0x6bc): undefined reference to `write1ByteTxRx(int, int, unsigned char, unsigned short, unsigned char)'
sendi_subscriber.cpp:(.text+0x6e4): undefined reference to `write4ByteTxRx(int, int, unsigned char, unsigned short, unsigned int)'
sendi_subscriber.cpp:(.text+0x6fe): undefined reference to `read4ByteTxRx(int, int, unsigned char, unsigned short)'
sendi_subscriber.cpp:(.text+0x720): undefined reference to `write1ByteTxRx(int, int, unsigned char, unsigned short, unsigned char)'
sendi_subscriber.cpp:(.text+0x72a): undefined reference to `closePort(int)'

help me

saufy

routiful, 2018-06-29:
Hi :)

I think that this issue is related [Dynamixel SDK](https://github.com/ROBOTIS-GIT/DynamixelSDK).
Next time you are supposed to open related issue on Issue page in Dynamixel SDK repo.

Dynamixel SDK has supported C++ in ROS Kinetic. So, you are supposed to use it.
Attached message is caused by fault setting in CMakeLists. 
Please follow below references and modified files.

http://wiki.ros.org/catkin/CMakeLists.txt

Thanks

robotpilot, 2018-07-05:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-07-05 by robotpilot

msaufyrohmad, 2018-07-05:
Yes
 This issue is solved.  Im sorry for no respond.  Thanks

Saufy