ROBOTIS-GIT/dynamixel-workbench #161: convertValue2Radian: Wrong conversion from ticks to radian

Martin-Oehler, 2018-06-29:
In the function convertValue2Radian defined [here](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_driver.h#L115), motor ticks are converted to radians. This is done by mapping the ticks of one revolution to 2*pi. 

Unfortunately in the [config file](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_item.cpp) the ticks are always one less than intended for a full revolution e.g. 4095 ticks instead of 4096 ticks for an MX-motor. This results in a wrong mapping between ticks and radian. This becomes especially noticeable if the motor rotates in the same direction for a long time as the error adds up.

We noticed this behavior when actuating our lidar with a dynamixel. After a while the reported radian position does not match the actual motor position anymore. We fixed this on our fork, so this ticket should make you aware of the bug.

Kind regards,
Martin Oehler

routiful, 2018-07-02:
Thank you for your help! I will amended it as soon as possible.

routiful, 2018-07-05:
I updated all value of max radian position and It is merged to master branch. (https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/162/commits/970fe330029eab5eeeaa938de4ccdc4fceecf962)

Thank you to your contribution and  if you have any issues or PRs, please feel free to open it any time.

closed 2018-07-05 by routiful