ROBOTIS-GIT/dynamixel-workbench #163: position_control

abhayvar, 2018-07-06:
ISSUE TEMPLATE ver. 0.1.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/)

1. Which distro of ROS have you installed? (ex, indigo, kinetic, melodic)
kinetic

2. Which OS have you installed? (ex, Ubuntu 16.04 LTS (Xenial Xerus), Linux Mint 18.x, ...)
ubuntu16.04

3. Which Dynamixel have you used? (ex, MX-28, XM430-W350, ...)
mx28,mx64 and ax18
 
4. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers position_control.launch
  ```
 
5. Copy and Paste your error message on terminal

 ... logging to /home/ayush/.ros/log/f277cba2-815c-11e8-879d-701ce7ad5ab6/roslaunch-Ayush-19846.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Ayush:46137/

SUMMARY
========

PARAMETERS
 * /baud_rate: 57600
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /scan_range: 20

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

ROS_MASTER_URI=http://localhost:11311

process[position_control-1]: started with pid [19863]
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : AX-18A
ID      : 1

MODEL   : AX-18A
ID      : 2

MODEL   : AX-18A
ID      : 3

MODEL   : AX-18A
ID      : 4

MODEL   : AX-18A
ID      : 5

MODEL   : AX-18A
ID      : 6

MODEL   : AX-18A
ID      : 7

MODEL   : MX-64
ID      : 8

MODEL   : MX-64
ID      : 9

MODEL   : MX-28
ID      : 10

MODEL   : MX-28
ID      : 11

MODEL   : MX-28
ID      : 12

MODEL   : MX-28
ID      : 13

MODEL   : MX-28
ID      : 14

MODEL   : MX-28
ID      : 15

MODEL   : MX-64
ID      : 16

MODEL   : (null)
ID      : 18

MODEL   : AX-18A
ID      : 19

-----------------------------------------------------------------------
================================================================================REQUIRED process [position_control-1] has died!
process has died [pid 19863, exit code -11, cmd /home/ayush/catkin_ws/devel/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/ayush/.ros/log/f277cba2-815c-11e8-879d-701ce7ad5ab6/position_control-1.log].
log file: /home/ayush/.ros/log/f277cba2-815c-11e8-879d-701ce7ad5ab6/position_control-1*.log
Initiating shutdown!
================================================================================
[position_control-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  
6. Please, describe detailedly what difficulty you are in 

    - We are using these 4mx64,6mx28 and 8ax18 all have 57600 baudrate. But after launching position_control it first launches, fixes the servos and shutdown, before this we did not added ax18 and at that time it does not shut down after launch, now becuase it shut down I can not publish angles to servo

routiful, 2018-07-09:
Hi :)

Would you try to launch it with only AX-18? Or change all Dynamixels from 57600 to 1M baudrate

abhayvar, 2018-07-09:
With only Ax18 they are working well.
But using them along with mx servos, the launch file kills

abhayvar, 2018-07-09:
And behaviour after changing baudrate is same

abhayvar, 2018-07-09:
And one more suspicious observation is Model Null, see the 18th place..instead of this we have a servo with id 17 but no servo with 18 id

abhayvar, 2018-07-09:
I mistakenly close this issue. I'm sorry for this, Pls let Routiful informed about my comments

routiful, 2018-07-10:
I set maximum 16 array size for Dynamixel ID storage. 

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/include/dynamixel_workbench_controllers/position_control.h#L54

If you change this more bigger, you can watch your Dynamixel has ID 16 or more.

abhayvar, 2018-07-10:
I changed that value to 20, still it does not solve anything

routiful, 2018-07-11:
I tested twenty AX-12 and two MX-28 with `position_control.launch`.
I attach result below

```
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : AX-12A
ID      : 1

MODEL   : AX-12A
ID      : 2

MODEL   : AX-12A
ID      : 4

MODEL   : AX-12A
ID      : 6

MODEL   : AX-12A
ID      : 7

MODEL   : AX-12A
ID      : 8

MODEL   : AX-12A
ID      : 9

MODEL   : MX-28
ID      : 10

MODEL   : MX-28
ID      : 11

MODEL   : AX-12A
ID      : 12

MODEL   : AX-12A
ID      : 13

MODEL   : AX-12A
ID      : 14

MODEL   : AX-12A
ID      : 15

MODEL   : AX-12A
ID      : 16

MODEL   : AX-12A
ID      : 17

MODEL   : AX-12A
ID      : 18

MODEL   : AX-12A
ID      : 21

MODEL   : AX-12A
ID      : 22

-----------------------------------------------------------------------
```

All Dynamixels are scanned and it moved by rosservice call command.
I only modified array size in `position_control.h`
I recommend you to double check id, baudrate and latest version of dynamixel firmware.

abhayvar, 2018-07-11:
I changed id 19 to 18 and that worked, I think the missing one id was creating the problem,
One more thing, to move these servos simultaneously I publish goal dynamixel position, but even after recording bites of position and giving that in input, the servo are not going to that position.
I think there is some error in other parameters of goal_dynamixel_position.
Can you please give an example of that(such that I can give position of a state recorded. 
It will be very thankful

routiful, 2018-07-12:
Sorry, I don't understand `recording bites of position`.
Would you explain what this mean?

abhayvar, 2018-07-12:
I mean position data i.e. 0-4095(mx64) and 0-1023(ax18)
We can echo this data, now how can we give the recorded position using goal position?
I need an example as in our case when i publish the data of a state(obtained through echo) but servos don't acheive the same position

routiful, 2018-07-13:
You can use `rostopic pub` commands like below example

```
rostopic pub /goal_dynamixel_position sensor_msgs/JointState "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
name: ['']
position: [0]
velocity: [0]
effort: [0]"
```
or make your own publisher and make message with [sensor_msgs/JointState](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/JointState.html) type

routiful, 2018-07-16:
Do you mean that you want to make profile to control Dynamixel?
Then, you set zero to `profile_velocity` and `profile_acceleration` parameters in launch file.

closed 2018-07-22 by robotpilot