ROBOTIS-GIT/dynamixel-workbench #174: single_manager.launch example fails with 12 servos

dshinsel, 2018-08-15:
ISSUE TEMPLATE ver. 0.1.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/)

1. Which distro of ROS have you installed? (ex, indigo, kinetic, melodic)
kinetic

2. Which OS have you installed? (ex, Ubuntu 16.04 LTS (Xenial Xerus), Linux Mint 18.x, ...)
Ubuntu 16.04 LTS

3. Which Dynamixel have you used? (ex, MX-28, XM430-W350, ...)
MX-106 + MX-64 + MX-28
 
4. Write down the commands you used in order
roslaunch my_dynamixel_workbench_tutorial single_manager.launch

single_manager.launch file:
<launch>
 <arg name="use_ping"         default="false"/>
 <arg name="id"               default="1"/>

 <arg name="device_name"      default="/dev/ttyUSB1"/>
 <arg name="baud_rate"        default="1000000"/>

 <arg name="scan_range"       default="200"/>

 <param name="ping"             value="$(arg use_ping)"  type="bool"/>
 <param name="ping_id"          value="$(arg id)"/>

 <param name="device_name"      value="$(arg device_name)"/>
 <param name="baud_rate"        value="$(arg baud_rate)"/>

 <param name="scan_range"        value="$(arg scan_range)"/>

 <node name="single_dynamixel_monitor" pkg="dynamixel_workbench_single_manager"
         type="single_dynamixel_monitor" required="true" output="screen"/>

 <node name="single_dynamixel_controller" pkg="dynamixel_workbench_single_manager"
         type="single_dynamixel_controller" required="true" output="screen"/>
 </launch>

5. Copy and Paste your error message on terminal
system@sheldon0:~/catkin_robot/src/my_dynamixel_workbench_tutorial/launch$ roslaunch my_dynamixel_workbench_tutorial single_manager.launch
... logging to /home/system/.ros/log/ac2d866e-a0c4-11e8-a5eb-a0c58952bec1/roslaunch-sheldon0-4958.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sheldon0:46385/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB1
 * /ping: False
 * /ping_id: 1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /scan_range: 200

NODES
  /
    single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
    single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

auto-starting new master
process[master]: started with pid [4968]
ROS_MASTER_URI=http://localhost:11311
[... 240 lines cut ...]
================================================================================
[single_dynamixel_controller-3] killing on exit
[single_dynamixel_monitor-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  

=======================================
6. Please, describe detailedly what difficulty you are in 

My robot has 19 servos, mix of MX106, RX64, MX28.
WORKS:  If I remove both arms (remaining servo count 7, mix of all above servos)
FAILS: if I reattach one arm, bringing servo count to 12 servos

NOTE:  Robot currently works with these packages:
   http://wiki.ros.org/dynamixel_motor
   http://wiki.ros.org/dynamixel_controllers
I wish to replace these packages and use packages from Robotis instead!

Thanks,
Dave Shinsel
www.dshinsel.com. Robot name: Sheldon

dshinsel, 2018-08-17:
Thanks, I guess that's what's intended by the name "Single Manager".

Funny, it sort-of works with a few servos attached, but fails if you have too many.

It would be REALLY helpful to have a utility to view all servos attached to a robot, and select / change each servo as needed, like exists on the excellent Windows application.

Thanks!

dshinsel, 2018-08-17:
One more related question.  Robotis has many packages, and I'm not clear f what I am looking for exists.  I want to control my robot arm, with many Robotis Servos, from ROS MoveIt.

I see "Open Manipulator" allows Moveit to control the Open Manipulator arm, but the code seems to be hard-coded to just 4 joints plus gripper, and the joint names are hard-coded to "joint1", etc.

Does Robotis have a package (or set of packages) to replace the old "dynamixel_controllers" and 
"arbotix" controllers?  I'd really like to switch from these older packages to more modern code that fully takes advantage of Robotis servos capabilities!

Is there something that exists like Open Manipulator that is more general-purpose and can be configured for any arm?

routiful, 2018-08-20:
We have provide `DYNAMIXEL WORKBENCH` is based on `DYNAMIXEL SDK`.
User can use this when they want to manage their Dynamixel and this software supports entire Dynamixel series includinf Dynamixel, Dynamixel X and Dynamixel PRO.

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

Futhermore, `OpenManipulator` is example to represent how to make manipulator using Dynamixels and
we provide some packages to control it.

http://emanual.robotis.com/docs/en/platform/openmanipulator/

If you have any ideas or suggestions about these packages, please feel free to comment tome :)

robotpilot, 2018-08-26:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-26 by robotpilot

AnastasiyaRybakova, 2019-11-26:
Hello, everyone. 
My name is Anastasiya. I am beginner in the robotics and programming, but I am doing my best.
Nowadays, I am working with robot arms. There are two. Each of them has 6 Dynamixel motors includes the gripper.
I am following instructions from here (http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#controllers ). 
My task is to control arms, which each joint is motor. I have to control them all together at the same moment for motions and body expressions. So I modified basic.yaml file and joint2.0. yaml file in the Dynamixel_workbench package and also tried to modify the motion.yaml file includes one arm's joint.
BUt when I run the command : roslaunch dynamixel_workbench_operators joint_operator.launch , there is the error like this:
[image]

My question is: 

- how can I fix this error?

- how can I control the whole arm which includes 6 dynamixels through ROS ?

Thank you, guy.