ROBOTIS-GIT/dynamixel-workbench #175: Segmentation fault (core dumped) after fresh install

Achllle, 2018-08-23:
## Settings
1. ROS kinetic
2. Ubuntu 16.04 LTS (Xenial Xerus)
3. Dynamixel model: XM540-W270-R

## Steps to reproduce
I followed the e-manual to install the workbench. I tried both installing from source and through debian package. (`catkin_make` when installing from source was successful and I sourced the workspace).

When I do 

```rosrun dynamixel_workbench_single_manager_gui dynamixel_workbench_single_manager_gui```

however I get the following error:

```Segmentation fault (core dumped)```

routiful, 2018-08-27:
Hi :)

You are supposed to open single_manager.launch before launch gui.

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#gui

Achllle, 2018-08-27:
Ah, did not see that. Thanks for the clarification. Perhaps could the error be replaced with something more sensible, like:

```
[ERROR]: single_manager node needs to be running. 
Did you launch single_manager.launch from dynamixel_workbench_single_manager?
```

closed 2018-08-27 by Achllle

routiful, 2018-08-27:
Thanks! I will add that warning messages next updates :)